ros-perception/image_common #33: Failure with distortion parameters

hbadino, 2015-05-22:
Hi,

the converter fails when converting from ini to yaml when the calibration was performed with k > 3 (k is the number of distortion parameters to optimize).

The error looks like:

[ERROR] [1432321037.860901266]: Failed to load camera model from file left.ini

When removing the additional distortion parameters from the ini file, it works just fine.

Hernan

vrabaud, 2015-06-20:
Sorry, been busy lkately. Do you have an .ini file around ? Thx

hbadino, 2015-06-30:
Thanks for replying. I do have a file... Try this one (I call it left.ini)

----

[image]

width
640

height
480

[mono_left]

camera matrix
369.344588 0.000000 320.739078
0.000000 367.154330 203.592450
0.000000 0.000000 1.000000

distortion
0.189544 -0.018229 -0.000630 0.000054 -0.000212 0.543582 -0.027892 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
262.927429 0.000000 320.984481 0.000000
0.000000 302.056213 188.592437 0.000000
0.000000 0.000000 1.000000 0.000000

----

My output is $ rosrun camera_calibration_parsers convert left.ini left.yaml
[ERROR] [1435679519.174873278]: Failed to load camera model from file left.ini

If instead, I leave only 5 fields in distortion (e.g., delete the last three numbers), then it works just fine.

Hernan

commit 98448a15 (2015-07-02): parse distortion of arbitraty length in INI
This fixes #33
closed 2015-07-02 by vrabaud