ros-perception/image_common #37: camera_info - image sync bug

antonioom, 2015-07-09:
I've noticed that the sequence numbers of the camera_info and image topcis advertised by an image_transport::CameraPublisher get out of sync when one subscribes to the camera_info topic but not to the image topic. This does not occur when one subscribes to the image topic topic but not to the camera_info. The sequence number should be incremenent when either topic or both topics are subscribed to (not only when the image topic is subscribed to).

The bug is easy to reproduce using rostopic echo /camera/image (or /camera/camera_info).

vrabaud, 2015-07-16:
```seq``` of the ROS messages headers is not present in the code of image_transport. Plus the code publishes both: https://github.com/ros-perception/image_common/blob/hydro-devel/image_transport/src/camera_publisher.cpp#L130
I'd say this is in ROS. What do you need the sequence ID for ?

jack-oquin, 2015-07-16:
Why does this matter? Are you just checking to verify nothing got lost?

The correct test would be for identical time stamps.

antonioom, 2015-07-17:
Yes, I thought of using seq numbers to issue warning messages when frames got dropped because a node was not able to keep up with the incoming data flow. (It was processing multiple video streams.) I was trying to debug a related problem and thought the seq numbers were involved in the problem. Thanks to your comments I figured out the problem lies somewhere else.

closed 2015-07-17 by antonioom

shubhamnagla, 2022-08-26:
@antonioom Hello,
I have been trying out a project which uses image_transport package. 
I keep on getting the following error:
  [image_transport] Topics '/camera/image_raw' and '/camera_info' do not appear to be synchronized. In the last 10s
 I am currently on 
 Ubuntu 22.04 
 ROS Humble
 Can you help me with some solution??

ijnek, 2022-08-26:
@shubhamnagla
This is an issue that was closed 7 years ago, you should rather go over to [ROS Answers](https://answers.ros.org/questions/) and post your question there and others might be able to help you out.