ros-perception/image_common #39: I got some weird error..

caemoon2, 2015-07-27:
when I tried to run following commad,
rosrun  camera_calibration_parsers convert  ost.ini cal.yml  

I got the result like this,
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::create_directory: No such file or directory
Aborted (core dumped)

and I don't know what happen and how to fix it.... I google for an hour to find any similar issues but I couldn't find anything and I"m really beginner of ROS and Linux.... So could you anybody help me to solve this problem?

vrabaud, 2015-07-28:
can you please send on my personal email, the ost.ini, thx.

caemoon2, 2015-07-28:
I sent you an email with my ost.ini file. please check it.

vrabaud, 2015-07-28:
thx for helping me out. I will release the fix now

commit e4b242c1 (2015-07-28): fix #39
closed 2015-07-28 by vrabaud