ros-perception/image_common #42: camera_info_manager: don't treat catkin_INCLUDE_DIRS as system-dir

kb0n, 2015-11-27:
I was installing ROS indigo (desktop-full) on a Fedora23 machine, following the "Installing from source" guide. I got the following error for the camera_info_manager package (full output):

```make
[kb0n@kb0n-macfed catkin_ws]$ ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-std=gnu++03 -DCMAKE_C_FLAGS=-std=gnu++03 --install-space /opt/ros/indigo --pkg camera_info_manager
Base path: /home/kb0n/ros/catkin_ws
Source space: /home/kb0n/ros/catkin_ws/src
Build space: /home/kb0n/ros/catkin_ws/build_isolated
Devel space: /home/kb0n/ros/catkin_ws/devel_isolated
Install space: /opt/ros/indigo
Additional CMake Arguments: -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-std=gnu++03 -DCMAKE_C_FLAGS=-std=gnu++03
==> Processing catkin package: 'camera_info_manager'
==> Creating build directory: 'build_isolated/camera_info_manager'
==> Building with env: '/opt/ros/indigo/env.sh'
==> cmake /home/kb0n/ros/catkin_ws/src/image_common/camera_info_manager -DCATKIN_DEVEL_PREFIX=/home/kb0n/ros/catkin_ws/devel_isolated/camera_info_manager -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-std=gnu++03 -DCMAKE_C_FLAGS=-std=gnu++03 -G Unix Makefiles in '/home/kb0n/ros/catkin_ws/build_isolated/camera_info_manager'
-- The C compiler identification is GNU 5.1.1
-- The CXX compiler identification is GNU 5.1.1
-- Check for working C compiler: /usr/lib64/ccache/cc
-- Check for working C compiler: /usr/lib64/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/lib64/ccache/c++
-- Check for working CXX compiler: /usr/lib64/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/kb0n/ros/catkin_ws/devel_isolated/camera_info_manager
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo;/opt/ros/fuerte/share/catkin/cmake/Modules;/opt/ros/fuerte
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.10")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python2.7/site-packages/em.pyc
-- Using empy: /usr/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/kb0n/ros/catkin_ws/build_isolated/camera_info_manager/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- Boost version: 1.58.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kb0n/ros/catkin_ws/build_isolated/camera_info_manager
==> make -j2 -l2 in '/home/kb0n/ros/catkin_ws/build_isolated/camera_info_manager'
Scanning dependencies of target camera_info_manager
[ 25%] Building CXX object CMakeFiles/camera_info_manager.dir/src/camera_info_manager.cpp.o
[ 50%] Linking CXX shared library /home/kb0n/ros/catkin_ws/devel_isolated/camera_info_manager/lib/libcamera_info_manager.so
[ 50%] Built target camera_info_manager
Scanning dependencies of target unit_test
[ 75%] Building CXX object CMakeFiles/unit_test.dir/tests/unit_test.cpp.o
[100%] Linking CXX executable /home/kb0n/ros/catkin_ws/devel_isolated/camera_info_manager/lib/camera_info_manager/unit_test
CMakeFiles/unit_test.dir/tests/unit_test.cpp.o: In function `delete_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
unit_test.cpp:(.text+0x27f): undefined reference to `ros::console::print(ros::console::FilterBase*, log4cxx::Logger*, ros::console::levels::Level, std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
CMakeFiles/unit_test.dir/tests/unit_test.cpp.o: In function `expected_calibration()':
unit_test.cpp:(.text+0x8d31): undefined reference to `sensor_msgs::distortion_models::PLUMB_BOB[abi:cxx11]'
CMakeFiles/unit_test.dir/tests/unit_test.cpp.o: In function `ros::ServiceClient::deserializeFailed(std::exception const&)':
unit_test.cpp:(.text._ZN3ros13ServiceClient17deserializeFailedERKSt9exception[_ZN3ros13ServiceClient17deserializeFailedERKSt9exception]+0xf0): undefined reference to `ros::console::print(ros::console::FilterBase*, log4cxx::Logger*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)'
collect2: error: ld returned 1 exit status
CMakeFiles/unit_test.dir/build.make:121: recipe for target '/home/kb0n/ros/catkin_ws/devel_isolated/camera_info_manager/lib/camera_info_manager/unit_test' failed
make[2]: *** [/home/kb0n/ros/catkin_ws/devel_isolated/camera_info_manager/lib/camera_info_manager/unit_test] Error 1
CMakeFiles/Makefile2:1001: recipe for target 'CMakeFiles/unit_test.dir/all' failed
make[1]: *** [CMakeFiles/unit_test.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'camera_info_manager':
  Command '['/opt/ros/indigo/env.sh', 'make', '-j2', '-l2']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/kb0n/ros/catkin_ws/build_isolated/camera_info_manager && /opt/ros/indigo/env.sh make -j2 -l2

Command failed, exiting.
```

The CFLAGS/CXXFLAGS didn't make any difference. 
After intensive testing, it turns out that line 15 in ```CMakeLists.txt``` was the culprit:
```CMake
include_directories(SYSTEM ${catkin_INCLUDE_DIRS})
```
Removing ```SYSTEM``` fixed it for me.

jack-oquin, 2015-11-27:
Thanks for the report. 

I don't know why that ``SYSTEM`` is there. It's probably not necessary, but it seems to have been there for a long time, so I worry that something may have depended on it in the past. 

@vrabaud: That parameter appears to go back to the original catkinization. Do you recall what was it for?

vrabaud, 2015-11-29:
SYSTEM has nothing to do here. Then again, it should only remove compilation warnings.

I think the reason was elsewhere: you have a problem during linkage and that executable was indeed not linked properly in the CMake. So it's fixed. You probably saw the error because you have some C++11 around and that's not the case on Ubuntu. Thx for reporting ! Releasing now.

closed 2015-11-29 by vrabaud

kb0n, 2015-11-29:
That's why I didn't provide a quick patch and reported an issue, waiting for you to look at it.

My first attempt was actually linking unit_test against the catkin_libs, but that didn't resolve anything. I also added the ```-DCMAKE_CXX_FLAGS=-std=gnu++03 -DCMAKE_C_FLAGS=-std=gnu++03``` flags to avoid C++11. 
I'm tired of looking for the actual problem, so I'm just happy with "works for me now" :)
Thanks

vrabaud, 2015-11-29:
Your GCC is also much more recent: Wily Werewolf builds that will happen with ROS Kinetic will help catch that too I hope.
Thx again ! (builds triggered btw).

commit 3ff494fa (2015-11-29): fix compilation on Fedora, fixes #42