ros-perception/image_common #49: Compressed image transport segmentation fault with OpenCV.

ethlu, 2016-03-08:
See issue here: http://answers.ros.org/question/228385/compressed_image_transport-segmentation-fault/

source code is this http://wiki.ros.org/image_transport/Tutorials/SubscribingToImages

Client side:
OS: Ubuntu 14.04
ROS Indigo
OpenCV version 3.1.0

Server side:
Raspberrypi Jessie
ROS Indigo

Here is a sample bag file of compressed video
[camera_compressed.bag.tar.gz](https://github.com/ros-perception/image_common/files/162511/camera_compressed.bag.tar.gz)

solosuper, 2016-03-10:
Hi, see my [workaround](https://github.com/ros-perception/image_common/issues/50#issuecomment-194903567) for this issue in #50.

vrabaud, 2016-03-11:
Is the OpenCV a custom version you built yourself ? This is not supported in Indigo (see REP 3) and could lead to whatever problem like crash. If you installed the ORS package, then it is irrelevant as it is not used by the official packages and you might as well remove it.
The image_transport plugins have never been tested with OpenCV3 anyway.

vrabaud, 2016-03-11:
what's the workaround you mention ? Compiling from source using OpenCV3 ? Well yes, that could work. You cannot use a package compiled with OpenCV2 with an installed OpenCV3, but if you recompile everything from source, that could work.

solosuper, 2016-03-11:
I have `ros-indigo-opencv3` installed, no custom built version. And yes, cloning `vision_opencv` and `image_transport_plugins` into the workspace and building with `catkin_make` solves it.

vrabaud, 2016-03-11:
well, I just want to make sure that this issue is normal: you are linking code to OpenCV3 while also linking it to a library (image_Transport) that is linked to OpenCV2.
This is C++, if you're linking to the same library with two different versions, the result is undefined. In the case of OpenCV, symbols changed a lot between the two so it makes sense it crashes when calling OpenCV (and that only happens for non-raw format).
If you agree, please close this issue and answer you question as resolved. Thx

ethlu, 2016-03-11:
I tried to build my own vision_opencv and image_transport_plugins. It still gives me segmentation fault. vrabaud: Is it something I can fix easily? Thanks.

solosuper, 2016-03-11:
Have you deleted `devel` and `build` folders before running `catkin_make`? My only suggestion.

ethlu, 2016-03-12:
I solved it by using find_package(OpenCV 2) instead of find_package(OpenCV) in my client code CMakeLists. Thanks everyone

closed 2016-03-12 by ethlu