ros-perception/image_common #52: image_transport republish compressed arm error

eybee, 2016-06-16:
I get the following error message when trying to to republish a raw image topic as a compressed one:

`rosrun image_transport republish raw in:=/usb_cam/image_raw compssed out:=/usb_cam/image`

`terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  According to the loaded plugin descriptions the class image_transport/compressed_pub with base class type image_transport::PublisherPlugin does not exist. Declared types are  image_transport/raw_pub
`

I'm running Ubuntu 14.04 arm with ROS Indigo on a raspberry pi 2.
The exact same command on a standard PC works as expected.

I know that normally the usb_cam package should output a compressed image topic by default because it uses the image_transport plugin system. However on the raspberry it doesn't. Maybe this is caused by the same issue that causes the error above.

eybee, 2016-06-17:
I solved it by installing the following packet:

`sudo apt-get install ros-indigo-compressed-image-transport`
Which is not included in the ros basic setup.

It would be nice though if the error message would suggest to install this package.

closed 2016-06-17 by eybee

jack-oquin, 2016-06-17:
> It would be nice though if the error message suggested to install this package.

Yes, it would. A pull request to improve that message might help.