ros-perception/image_common #53: Customize set_camera_info service URL?

chrisspen, 2016-08-23:
How do you customize the URL that that set_camera_info service is published at via the CameraInfoManager?

I'm attempting to patch the [unmaintained raspicam ROS node](https://github.com/fpasteau/raspicam_node) so I can calibrate it using the camera_calibration node. However, by default it publishes its info service to `/set_camera_info` instead of `/camera/set_camera_info' so when I run the calibration process, it errors out with:

    ('Waiting for service', '/camera/set_camera_info', '...')
    Service not found

I've searched through [the docs](http://docs.ros.org/api/camera_info_manager/html/classcamera__info__manager_1_1CameraInfoManager.html) but I can't find any option for setting this path.

jack-oquin, 2016-08-24:
The [CameraInfoManager constructor](http://docs.ros.org/kinetic/api/camera_info_manager/html/camera__info__manager_8cpp_source.html#l00078) advertises ``set_camera_info`` service in the namespace of whatever node handle you pass it.

Is the raspicam node already running in a ``/camera`` namespace?

chrisspen, 2016-08-24:
I believe so, and `rostopic list` shows it exposes `/camera/image`, but I only see `/set_camera_info`.

Is there a sequence when CameraInfoManager should be initialized? The node is currently doing right after `ros::init()` as [shown here](https://github.com/fpasteau/raspicam_node/blob/master/src/raspicam_raw_node.cpp#L822). Is that too early?

I've forked the code and have fixed a few minor bugs, so if this isn't the correct initialization, I could fix it.

jack-oquin, 2016-08-24:
Sounds like you're *not* running in the ``/camera`` namespace, just [advertising the topic with a relative topic name](https://github.com/fpasteau/raspicam_node/blob/master/src/raspicam_node.cpp#L804).

If you want things to be in the ``/camera`` namespace, you should probably define a camera node handle, [like in this example](https://github.com/ros-drivers/camera1394/blob/master/src/nodes/camera1394_node.cpp#L64). Use that for both the topics and the camera info service.

chrisspen, 2016-08-24:
I see. I'll try updating the code following that example. Thanks.

jack-oquin, 2016-08-24:
Note that the example given does *not* follow the recommendations of [REP-0135](http://www.ros.org/reps/rep-0135.html), which recommends *not* using the "camera" sub-namespace, but rather launching the driver in whatever namespace seems appropriate for that specific device.

I chose not to modify camera1394 when that REP was adopted, because it already had many users and I didn't want to confuse them. Making that change in a backwards-compatible way seemed difficult.

I would recommend using the root namespace, in your case.

chrisspen, 2016-08-24:
What do you mean by root namespace? As in "/"? Shouldn't I use something like "raspicam"?

jack-oquin, 2016-08-24:
No, the ROS root namespace is ``/``. Using ``raspicam`` makes the data specific to that one driver. The general interface is ``/image_raw``, which works with any camera. not just yours.

chrisspen, 2016-08-25:
I see what you mean. So it's actually correct the way it is, exposing the service at `/set_camera_info`, because all I need to do is launch the node inside a "camera" namespace to make it appear as "/camera/set_camera_info", and I should remove the hard-coded "/camera" so the other publishers will work the same way.

closed 2016-08-25 by chrisspen