ros-perception/image_common #62: Missing catkin-dependency from info-manager to calibration-parsers

mightyCelu, 2017-05-03:
The camera_info_manager includes and links against libraries from camera_calibration_parsers. However, there is no corresponding dependency added to the catkin_package-call.
Adding this dependency caused a linker error in a dependent package (cv_camera) with my cross-compilation build.

jack-oquin, 2017-05-04:
Thanks for the report. I believe you are correct.

Are you saying there's a problem with adding the desired CATKIN_DEPENDS to ``catkin_package()``?

mightyCelu, 2017-05-04:
No, this is what fixed the problem for me. If you'd like I can provide a PR. However I am not that used to working with catkin so I might be overlooking some side-effects of my solution.

closed 2017-05-05 by vrabaud