ros-perception/image_common #64: Several camera info managers

arnasIV, 2017-05-10:
If I create two camera info managers in the same place of my program/ in the same node and I want to be able to set camera info for each of two cameras (left and right), i. e. use service left/set_camera_info and right/set_camera_info, how do I add this /left or /right namespace?

ros::NodeHandle &nh = getNodeHandle();
camera_info_manager::CameraInfoManager(nh, left_name, left_camera_info_url) left;
camera_info_manager::CameraInfoManager(nh, right_name, right_camera_info_url) right;

jack-oquin, 2017-05-31:
Define node handles for the two subordinate name spaces.

This question would be better asked on answers.ros.org.

jack-oquin, 2017-05-31:
We try to restrict issues to bugs and enhancement requests.

closed 2017-05-31 by jack-oquin

arnasIV, 2017-05-31:
Sorry, but I tried and it did not work.