ros-perception/image_common #70: image_transport::ImageTransport throwing *** stack smashing detected ***

nathangeorge1, 2018-01-25:
Hello,

When used Ubuntu Software Manger to update, it appears to have updated ROS. I know for a separate issue there's a certain .so file that was updated which is breaking something unrelated.

For this issue, if I simply try to run `	image_transport::ImageTransport it(nh);`, I get the following error:

`*** stack smashing detected ***: my/path/to/executable terminated`

My guess is something is wrong internally with ImageTransport; what should I do?

tfoote, 2018-01-25:
If you did a partial update you might have pulled a [recent incompatible ABI change](https://discourse.ros.org/t/new-packages-for-indigo-2017-12-26/3541/2?u=tfoote) In general the solution to this is to make sure to do a full `apt-get dist-upgrade` or use the Ubuntu Software Manager to make sure that you have the latest of all the ROS packages. 

To help you more I'll need more information like what rosdistro you're running what packages you updated as well as the versions your now running. 

If you can create a small reproducible program that will help isolate the problem.

nathangeorge1, 2018-01-26:
Hello,

I did use Ubuntu Software Manager, but it's what caused this error. I also just tried `apt-get dist-upgrade` which downloaded 300MB of stuff, but it's still broken.

I am running rosdistro kinetic. I'm not sure how to find what packages I updated using Software Manager, as that happened yesterday; any suggestions? When using `apt-get dist-upgrade`, it updated the following:

```
The following NEW packages will be installed:
  linux-headers-4.13.0-32 linux-headers-4.13.0-32-generic linux-image-4.13.0-32-generic
  linux-image-extra-4.13.0-32-generic linux-signed-image-4.13.0-32-generic
The following packages will be upgraded:
  fwupdate fwupdate-signed libfwup0 libtasn1-6 linux-generic-hwe-16.04 linux-headers-generic-hwe-16.04
  linux-image-generic-hwe-16.04 linux-signed-generic-hwe-16.04 linux-signed-image-generic-hwe-16.04
9 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
```

Here is the small reproducible program causing the bug:

```
// General Includes
#include <stdlib.h>
#include <string>
#include <iostream>
#include <stdio.h>
#include <vector>
#include <signal.h>

// ROS Includes
//#include "ros/ros.h"
#include <image_transport/image_transport.h>

// OpenCV Includes
#include <opencv2/core.hpp>

// Namespaces
using namespace std;
using namespace cv;

void shutdownIntegrationTrial(int signal) {
	ros::shutdown();
}

/** @file
 *  @brief Integration Trials
 */
int main( int argc, char** argv ) {
	// init
	ros::init(argc, argv, "integration_trial");
	ros::NodeHandle nh;
	signal(SIGINT, shutdownIntegrationTrial);

	image_transport::ImageTransport it(nh);

	// end
	ROS_INFO("Ended and Spinning");
	while(ros::ok()) ros::spinOnce();
	return EXIT_SUCCESS;
}
```

Which is causing this bug:

```
*** stack smashing detected ***: path_to_repo/devel/lib/plant_trimmer/IntegrationTester terminated
```

nathangeorge1, 2018-01-26:
I was able to find the Software Manager updates log (at /var/log/apt/history.log); I believe this is the long update that Software Manager did:

```
Start-Date: 2018-01-25  10:26:18
Commandline: aptdaemon role='role-commit-packages' sender=':1.69'
Install: python-wxgtk3.0:amd64 (3.0.2.0+dfsg-1build1, automatic), libwxgtk3.0-0v5:amd64 (3.0.2+dfsg-1.3ubuntu0.1, automatic), libllvm5.0:amd64 (1:5.0-3~16.04.1, automatic), python-wxtools:amd64 (3.0.2.0+dfsg-1build1, automatic), libwxbase3.0-0v5:amd64 (3.0.2+dfsg-1.3ubuntu0.1, automatic), libtinyxml2-dev:amd64 (2.2.0-1.1ubuntu1, automatic), python-wxversion:amd64 (3.0.2.0+dfsg-1build1, automatic), libdrm-common:amd64 (2.4.83-1~16.04.1, automatic)
Upgrade: ca-certificates-java:amd64 (20160321, 20160321ubuntu1), ros-kinetic-interactive-markers:amd64 (1.11.3-0xenial-20170727-193453-0800, 1.11.3-0xenial-20180105-170022-0800), ros-kinetic-vision-opencv:amd64 (1.12.4-0xenial-20170613-183914-0800, 1.12.7-0xenial-20171117-073106-0800), ros-kinetic-image-pipeline:amd64 (1.12.20-0xenial-20170727-205957-0800, 1.12.22-0xenial-20180105-155130-0800), ros-kinetic-rqt-gui-py:amd64 (0.5.0-0xenial-20170613-192652-0800, 0.5.0-0xenial-20171116-230001-0800), libnm-glib4:amd64 (1.2.6-0ubuntu0.16.04.1, 1.2.6-0ubuntu0.16.04.2), ros-kinetic-rosparam:amd64 (1.12.7-0xenial-20170217-215616-0800, 1.12.12-0xenial-20171116-190414-0800), ros-kinetic-opencv3:amd64 (3.2.0-4xenial-20170607-095735-0800, 3.3.1-0xenial-20171106-101742-0800), python-rosdistro:amd64 (0.6.2-1, 0.6.6-100), ros-kinetic-urdf-parser-plugin:amd64 (1.12.11-0xenial-20170627-162844-0800, 1.12.12-0xenial-20171113-102356-0800), ros-kinetic-eigen-stl-containers:amd64 (0.1.8-0xenial-20170310-121851-0800, 0.1.8-0xenial-20171104-173958-0800), ros-kinetic-tf2-msgs:amd64 (0.5.16-0xenial-20170727-180809-0800, 0.5.17-0xenial-20180105-133131-0800), ros-kinetic-geometric-shapes:amd64 (0.5.2-0xenial-20170613-175238-0800, 0.5.3-1xenial-20171130-112938-0800), suru-icon-theme:amd64 (14.04+16.04.20161024-0ubuntu1, 14.04+16.04.20171116-0ubuntu1), ros-kinetic-rospy:amd64 (1.12.7-0xenial-20170613-172037-0800, 1.12.12-0xenial-20171116-210432-0800), ros-kinetic-desktop:amd64 (1.3.1-0xenial-20170915-160631-0800, 1.3.1-0xenial-20180105-194942-0800), ros-kinetic-roslaunch:amd64 (1.12.7-0xenial-20170613-174031-0800, 1.12.12-0xenial-20171116-224009-0800), ros-kinetic-roscpp-traits:amd64 (0.6.2-0xenial-20170217-215757-0800, 0.6.7-0xenial-20171104-173916-0800), python2.7-dev:amd64 (2.7.12-1ubuntu0~16.04.2, 2.7.12-1ubuntu0~16.04.3), ros-kinetic-rqt-tf-tree:amd64 (0.5.8-0xenial-20170727-184154-0800, 0.5.8-0xenial-20180105-150323-0800), ros-kinetic-rqt-moveit:amd64 (0.5.7-0xenial-20170613-200239-0800, 0.5.7-0xenial-20171117-045334-0800), ros-kinetic-visualization-marker-tutorials:amd64 (0.10.1-0xenial-20170613-174117-0800, 0.10.2-0xenial-20180105-010325-0800), libgles2-mesa:amd64 (17.0.7-0ubuntu0.16.04.2, 17.2.4-0ubuntu1~16.04.4), libseccomp2:amd64 (2.2.3-3ubuntu3, 2.3.1-2.1ubuntu2~16.04.1), libcrack2:amd64 (2.9.2-1build2, 2.9.2-1ubuntu1), ros-kinetic-roslint:amd64 (0.11.0-0xenial-20170217-225443-0800, 0.11.0-0xenial-20171104-175010-0800), ros-kinetic-ros-base:amd64 (1.3.1-0xenial-20170714-102652-0800, 1.3.1-0xenial-20171117-082743-0800), ros-kinetic-roslisp:amd64 (1.9.21-0xenial-20170714-101403-0800, 1.9.21-0xenial-20171116-190201-0800), python-catkin-pkg-modules:amd64 (0.3.8-1, 0.4.1-1), ros-kinetic-ros-comm:amd64 (1.12.7-0xenial-20170714-101801-0800, 1.12.12-0xenial-20171117-050801-0800), ros-kinetic-nodelet-topic-tools:amd64 (1.9.10-0xenial-20170613-185958-0800, 1.9.14-0xenial-20171117-052227-0800), update-manager-core:amd64 (1:16.04.9, 1:16.04.10), ros-kinetic-ros-core:amd64 (1.3.1-0xenial-20170714-102336-0800, 1.3.1-0xenial-20171117-082647-0800), libnux-4.0-common:amd64 (4.0.8+16.04.20160705-0ubuntu1, 4.0.8+16.04.20170816-0ubuntu1), ros-kinetic-rqt-console:amd64 (0.4.8-0xenial-20170613-200311-0800, 0.4.8-0xenial-20171117-050836-0800), unity-control-center-faces:amd64 (15.04.0+16.04.20170214-0ubuntu1, 15.04.0+16.04.20170214-0ubuntu2), ros-kinetic-tf2-ros:amd64 (0.5.16-0xenial-20170727-182021-0800, 0.5.17-0xenial-20180105-145343-0800), libdrm-nouveau2:amd64 (2.4.76-1~ubuntu16.04.1, 2.4.83-1~16.04.1), ros-kinetic-rospy-tutorials:amd64 (0.7.1-0xenial-20170613-182829-0800, 0.7.1-0xenial-20171116-225802-0800), ros-kinetic-rqt-pose-view:amd64 (0.5.7-0xenial-20170727-192003-0800, 0.5.8-0xenial-20180105-173747-0800), ros-kinetic-diagnostic-common-diagnostics:amd64 (1.9.2-0xenial-20170727-192538-0800, 1.9.2-0xenial-20180105-171351-0800), ros-kinetic-pcl-msgs:amd64 (0.2.0-0xenial-20170613-162935-0800, 0.2.0-0xenial-20171104-181550-0800), ros-kinetic-rqt-logger-level:amd64 (0.4.8-0xenial-20170613-195346-0800, 0.4.8-0xenial-20171117-050420-0800), ros-kinetic-topic-tools:amd64 (1.12.7-0xenial-20170613-182844-0800, 1.12.12-0xenial-20171116-225154-0800), ros-kinetic-roslib:amd64 (1.13.5-0xenial-20170308-181854-0800, 1.13.6-0xenial-20171116-190447-0800), ros-kinetic-diagnostic-msgs:amd64 (1.12.5-0xenial-20170613-161627-0800, 1.12.5-0xenial-20171104-180007-0800), ros-kinetic-rosgraph-msgs:amd64 (1.11.2-0xenial-20170613-161833-0800, 1.11.2-0xenial-20171104-181933-0800), ros-kinetic-robot-model:amd64 (1.12.11-0xenial-20170915-160033-0800, 1.12.11-0xenial-20180105-190034-0800), ros-kinetic-tf2-geometry-msgs:amd64 (0.5.16-0xenial-20170727-182525-0800, 0.5.17-0xenial-20180105-150108-0800), ros-kinetic-roslz4:amd64 (1.12.7-0xenial-20170308-182349-0800, 1.12.12-0xenial [... cut ...] .4.9-0xenial-20171117-051044-0800), ros-kinetic-webkit-dependency:amd64 (1.1.0-0xenial-20170416-172441-0800, 1.1.0-0xenial-20171104-172845-0800), dpkg-dev:amd64 (1.18.4ubuntu1.2, 1.18.4ubuntu1.3), qtermwidget59-data:amd64 (0.6.1-0xenial1, 0.6.1-0xenial2), ros-kinetic-tf2-py:amd64 (0.5.16-0xenial-20170727-181635-0800, 0.5.17-0xenial-20180105-145030-0800), ros-kinetic-gazebo-ros-pkgs:amd64 (2.5.13-0xenial-20170727-212737-0800, 2.5.14-1xenial-20180105-183345-0800)
End-Date: 2018-01-25  10:30:38
```

nathangeorge1, 2018-01-26:
Chris from ROS helped me fix this issue. There's a bug in tinyxml2 so when ROS updated; it didn't something weird to tinyxml2. I think it has something to do with static initialization, but I wasn't able to understand that part.

Anyways, the bug is fixed; thanks.

tfoote, 2018-01-26:
Glad it's working. Closing this issue.

closed 2018-01-26 by tfoote

clalancette, 2018-01-29:
I'll leave a quick note for some context on what I think was going on here.  We didn't debug down to an actual root cause because once we found out tinyxml2 was involved, @nathangeorge1 told me he didn't need it, so we just removed it and got rid of the problem.

In any case, what we ended up debugging it down to was that the executable that was being built was linked against the system version of `libtinyxml2.so`, and also had a self-vendored version of the `tinyxml2.cpp` file in the source tree.  I'm guessing that during startup, there was some static initialization that was being done twice (once in the main executable, once when the system `libtinyxml2` was being dynamically linked in), and it was causing the smashing problem.  One problem with this theory is that I don't quite understand why the linker didn't complain about duplicate symbols during link time.

In any case, as @nathangeorge1, he is good for now.

gaodafa, 2018-07-17:
hi, nathan, i got the same problem today. how did you solve yours? could you post it? many thx~

nathangeorge1, 2018-07-17:
Hi @gaodafa,

All I did was remove tinyxml2 from my project; apart from that I don't really know how to solve this. Chances are that it's a specific dependency in your CMakelists.txt that is improperly configured. Try removing your dependencies one by one until your ImageTransport object works.

gaodafa, 2018-07-31:
thank you, @nathangeorge1, the problem is solved by removing my local libtinyxml2.so and using the system version of libtinyxml2.so.