ros-perception/image_common #78: recent yaml-cpp releases require C++11 ==> compile error on macos

NikolausDemmel, 2018-03-30:
It seems recent yaml-cpp releases [require C++11](https://github.com/jbeder/yaml-cpp/releases/tag/yaml-cpp-0.5.3), leading to compile errors for `camera_calibration_parsers` on OSX, as the current version on homebrew is `0.6.2`.

Switching back to the previous version of yaml-cpp (which was still installed) works around the issue.

```
brew switch yaml-cpp 0.5.3
```

compile error:
```
[camera_calibration_parsers:make] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DHAVE_NEW_YAMLCPP -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"camera_calibration_parsers\" -Dcamera_calibration_parsers_EXPORTS -I/Users/demmeln/work/ros/ros-install-osx/lunar_desktop_full_ws/src/image_common/camera_calibration_parsers/include -I/opt/ros/lunar/include -I/opt/ros/lunar/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/usr/local/include -I/usr/local/Cellar/python@2/2.7.14_3/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/usr/local/Cellar/yaml-cpp/0.6.2/include  -O3 -DNDEBUG -fPIC   -o CMakeFiles/camera_calibration_parsers.dir/src/parse_yml.cpp.o -c /Users/demmeln/work/ros/ros-install-osx/lunar_desktop_full_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp
[camera_calibration_parsers:make] In file included from /Users/demmeln/work/ros/ros-install-osx/lunar_desktop_full_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:38:
[camera_calibration_parsers:make] In file included from /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/yaml.h:10:
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/parser.h:41:3: warning: explicit conversion functions are a C++11 extension [-Wc++11-extensions]
[camera_calibration_parsers:make]   explicit operator bool() const;
[camera_calibration_parsers:make]   ^~~~~~~~
[camera_calibration_parsers:make] In file included from /Users/demmeln/work/ros/ros-install-osx/lunar_desktop_full_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:38:
[camera_calibration_parsers:make] In file included from /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/yaml.h:14:
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:100:47: error: a space is required between consecutive right angle brackets (use '> >')
[camera_calibration_parsers:make]     const T&, typename disable_if<is_numeric<T>>::type* = 0) {
[camera_calibration_parsers:make]                                               ^~
[camera_calibration_parsers:make]                                               > >
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:112:50: error: a space is required between consecutive right angle brackets (use '> >')
[camera_calibration_parsers:make]     const T& key, typename enable_if<is_numeric<T>>::type* = 0) {
[camera_calibration_parsers:make]                                                  ^~
[camera_calibration_parsers:make]                                                  > >
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:123:11: error: exception specification of overriding function is more lax than base version
[camera_calibration_parsers:make]   virtual ~Exception() YAML_CPP_NOEXCEPT;
[camera_calibration_parsers:make]           ^
[camera_calibration_parsers:make] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/stdexcept:93:13: note: overridden virtual function is here
[camera_calibration_parsers:make]     virtual ~runtime_error() _NOEXCEPT;
[camera_calibration_parsers:make]             ^
[camera_calibration_parsers:make] In file included from /Users/demmeln/work/ros/ros-install-osx/lunar_desktop_full_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:38:
[camera_calibration_parsers:make] In file included from /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/yaml.h:14:
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:123:23: error: expected ';' at end of declaration list
[camera_calibration_parsers:make]   virtual ~Exception() YAML_CPP_NOEXCEPT;
[camera_calibration_parsers:make]                       ^
[camera_calibration_parsers:make]                       ;
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:125:33: warning: defaulted function definitions are a C++11 extension [-Wc++11-extensions]
[camera_calibration_parsers:make]   Exception(const Exception&) = default;
[camera_calibration_parsers:make]                                 ^
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:148:45: warning: defaulted function definitions are a C++11 extension [-Wc++11-extensions]
[camera_calibration_parsers:make]   ParserException(const ParserException&) = default;
[camera_calibration_parsers:make]                                             ^
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:149:29: error: expected ';' at end of declaration list
[camera_calibration_parsers:make]   virtual ~ParserException() YAML_CPP_NOEXCEPT;
[camera_calibration_parsers:make]                             ^
[camera_calibration_parsers:make]                             ;
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:156:61: warning: defaulted function definitions are a C++11 extension [-Wc++11-extensions]
[camera_calibration_parsers:make]   RepresentationException(const RepresentationException&) = default;
[camera_calibration_parsers:make]                                                             ^
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:157:37: error: expected ';' at end of declaration list
[camera_calibration_parsers:make]   virtual ~RepresentationException() YAML_CPP_NOEXCEPT;
[camera_calibration_parsers:make]                                     ^
[camera_calibration_parsers:make]                                     ;
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:165:41: warning: defaulted function definitions are a C++11 extension [-Wc++11-extensions]
[camera_calibration_parsers:make]   InvalidScalar(const InvalidScalar&) = default;
[camera_calibration_parsers:make]                                         ^
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:166:27: error: expected ';' at end of declaration list
[camera_calibration_parsers:make]   virtual ~InvalidScalar() YAML_CPP_NOEXCEPT;
[camera_calibration_parsers:make]                           ^
[camera_calibration_parsers:make]                           ;
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:175:37: warning: defaulted function definitions are a C++11 extension [-Wc++11-extensions]
[camera_calibration_parsers:make]   KeyNotFound(const KeyNotFound&) = default;
[camera_calibration_parsers:make]                                     ^
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:176:25: error: expected ';' at end of declaration list
[camera_calibration_parsers:make]   virtual ~KeyNotFound() YAML_CPP_NOEXCEPT;
[camera_calibration_parsers:make]                         ^
[camera_calibration_parsers:make]                         ;
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:184:30: error: expected ';' at end of declaration list
[... 52 lines cut ...]
[camera_calibration_parsers:make]   virtual ~EmitterException() YAML_CPP_NOEXCEPT;
[camera_calibration_parsers:make]                              ^
[camera_calibration_parsers:make]                              ;
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:260:29: warning: defaulted function definitions are a C++11 extension [-Wc++11-extensions]
[camera_calibration_parsers:make]   BadFile(const BadFile&) = default;
[camera_calibration_parsers:make]                             ^
[camera_calibration_parsers:make] /usr/local/Cellar/yaml-cpp/0.6.2/include/yaml-cpp/exceptions.h:261:21: error: expected ';' at end of declaration list
[camera_calibration_parsers:make]   virtual ~BadFile() YAML_CPP_NOEXCEPT;
[... cut ...]
[camera_calibration_parsers:make]                                           > >
[camera_calibration_parsers:make] fatal error: too many errors emitted, stopping now [-ferror-limit=]
[camera_calibration_parsers:make] 14 warnings and 20 errors generated.
[camera_calibration_parsers:make] make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_yml.cpp.o] Error 1
[camera_calibration_parsers:make] make[1]: *** [CMakeFiles/camera_calibration_parsers.dir/all] Error 2
[camera_calibration_parsers:make] make: *** [all] Error 2
```

smnogar, 2018-04-23:
Yep, I'm getting the same thing.

Tobias-Fischer, 2018-04-25:
Simply adding `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")` to `CMakeLists.txt` solves this issue. I cannot find a branch for `kinetic` and newer, where C++11 is a minimum requirement (http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021), so I can't / don't want to open a pull request for this.

NikolausDemmel, 2018-04-26:
@Tobias-Fischer, yeah, that should work around the issue, but it is still unclear to me what a "clean" solution is.

Firstly, adding this to the CMakeLists.txt has some issues:
 1. If you compile your workspace with catkin_cmake, that setting applies to all packages, if you use catkin_tools, it only applies to this one package.
 2. If you use a compiler that has C++14 or newer by default, this setting is a "downgrade".
 3. ...

Other unclear question: Do we need users of image_common to also compile with C++11, or is it fine for them to stick with an older standard?

Maybe the best approach would be to find a clean way to "compile the whole workspace with C++11" for OSX users and add that to the install instructions.

Some previous discussion:
 * https://answers.ros.org/question/246369/the-good-c1114-the-bad-cmake_cxx_flags-and-the-ugly-cmake/?answer=246418#post-id-246418
 * https://github.com/ros/robot_state_publisher/pull/78
 * https://github.com/ros-industrial/industrial_ci/issues/89
 * https://github.com/ros-planning/moveit/issues/289

NikolausDemmel, 2018-05-04:
For the record, I've had success with passing additionally setting -DCMAKE_CXX_STANDARD=14 in catkin config to compile lunar.