ros-perception/image_common #80: No name resolution of base topic in image_transport::Subscriber

yoshito-okada, 2018-04-19:
image_transport::Subscriber does not resolve the name of base topic although CameraSubscriber, CameraPublisher, and Publisher do.  This causes an issue that a base topic remapping does not work when the transport type is not `raw`.  For instance, the following settings subscribe not `camera/image_raw/compressed` but `image_in/compressed`.
```
<remap from="image_in" to="camera/image_raw"/>
<param name="image_transport" value="compressed"/>
```

christian-rauch, 2018-05-06:
I have a similar issue with the `SubscriberFilter` (https://answers.ros.org/question/290472/topic-remapping-in-synchronised-message_filters/). The full topic name gets compiled before the topic remapping is applied: https://github.com/ros-perception/image_common/blob/f9e705aac2218f0ba97fd029e9d7ba709d17dae4/image_transport/include/image_transport/simple_subscriber_plugin.h#L98-L101

bbferka, 2019-11-26:
I have the same issue on a multi camera system. I though I could simply remap in the launch file from a generic ``image`` topic to the actual camera topic and set the ``image_transport`` parameter to ``compressed`` but it just ended up subscribing to ``image/compressed`` and I end up having to remap this latter one instead.

NicoMandel, 2020-05-04:
I have the same issue and still no clue how to solve it. Does anyone have a solution yet @christian-rauch ?

christian-rauch, 2020-05-04:
The only workaround I found is to manually remap the topic:
```C++
ros::NodeHandle n_priv;

image_transport::SubscriberFilter sub_image_depth;

sub_image_depth.subscribe(
    it,
    ros::names::remap("depth/image"), // manually remap topic from 'depth/image' to target
    1,
    image_transport::TransportHints("raw", ros::TransportHints(), n_priv, "depth/image_transport")
);
```

See https://github.com/christian-rauch/depth_box_filter/blob/master/src/depth_box_filter_node.cpp for the full example.

NicoMandel, 2020-05-04:
Thank you for the advice @christian-rauch . I am still a little confused, I have a raspberry running as an onboard computer, with a marker detection package. Can I just apply the fix [in this line](https://github.com/NicoMandel/marker_localization/blob/80297693777c65044dccb979d31c3466a0ee1b9c/ml_detector/src/ml_detector_node.cpp#L148) and then continue with the namespace remapping as is or am I basically hard-coding the names of the image topics now, without a chance to remap?

christian-rauch, 2020-05-04:
> Can I just apply the fix [in this line](https://github.com/NicoMandel/marker_localization/blob/80297693777c65044dccb979d31c3466a0ee1b9c/ml_detector/src/ml_detector_node.cpp#L148) and then continue [...]

If you apply the remapping via `ros::names::remap("image_raw")` then you do not have to hardcode anything.

Just take the ROS node at https://github.com/christian-rauch/depth_box_filter as an example. The remapping is done in the launch file and the `remap` method provides this mapping in the node.

NicoMandel, 2020-05-05:
Somehow, after applying the fix, see [here](https://github.com/NicoMandel/marker_localization/blob/98d55dec257470a173c9f01948a3c852fd55b030/ml_detector/src/ml_detector_node.cpp#L148)
And using a launch file to try with my webcam, see [here](https://github.com/NicoMandel/marker_localization/blob/98d55dec257470a173c9f01948a3c852fd55b030/ml_detector/launch/webcam_detect.launch#L31) 
with the following lines:
```
        <remap from="~image_raw" to="$(arg image_topic)"/>
        <remap from="~camera_info" to="$(arg camera_info_topic)"/>
        <param if="$(arg compressed)" name="image_transport" value="compressed"/>
```
I still get the same issue, with the subscription going to:
```
Subscriptions: 
 * /ml_detector/image_raw/compressed [unknown type]
 * /webcam/camera_info [sensor_msgs/CameraInfo]
```
What am I missing here? I recompile and source the workspace...

christian-rauch, 2020-05-05:
Use my `depth_box_filter` as a template to verify that this approach works for you. This example contains the node with the remapping and a launch file, which you can use to reproduce this effect for your node. If the remapping in my nodes does not work, please report an issue report there. Your launch file differs from mine in that you use the private namespace (`~`).

Otherwise, this issue is not a discussion forum. If you have further troubles with remapping topic names, please ask this specific question at https://answers.ros.org.

KenYN, 2020-10-09:
Just a quick note for nodelets: you need to use `nodelet::Nodelet::getRemappingArgs()` for them to resolve this issue.