ros-perception/image_common #81: Issue compiling image_transport on Fedora 28

nordmoen, 2018-05-02:
I'm having some issues compiling image_transport version `1.11.13-0` on Fedora 28.

The problem is a long soup of template issues, but boils down to:
```
Repos/ros_catkin_ws/src/image_common/image_transport/src/camera_subscriber.cpp:112:64:   required from here
Repos/ros_catkin_ws/install_isolated/include/message_filters/synchronizer.h:358:14: error: invalid operands of types ‘<unresolved overloaded function type>’ and ‘int’ to binary ‘operator<’
     this->add<i>(evt);
```

I'm not sure what information is necessary to fix this issue. It seems that some flags to GCC has changed as I did not have any problem compiling image_transport on Fedora 27.

Parameters from CMake:
```
-- The C compiler identification is GNU 8.0.1
-- The CXX compiler identification is GNU 8.0.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX:  Repos/ros_catkin_ws/devel_isolated/image_transport
-- Using CMAKE_PREFIX_PATH:  Repos/ros_catkin_ws/install_isolated
-- This workspace overlays:  Repos/ros_catkin_ws/install_isolated
-- Found PythonInterp: /usr/bin/python (found version "2.7.14") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python2.7/site-packages/em.pyc  
-- Using empy: /usr/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR:  Repos/ros_catkin_ws/build_isolated/image_transport/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.11
-- Boost version: 1.66.0
-- Configuring done
-- Generating done
-- Build files have been written to: Repos/ros_catkin_ws/build_isolated/image_transport
```

nordmoen, 2018-05-02:
Ok so the problem doesn't seem to be in image_transport, but in [`ros_comm/message_filters/synchronizer.h`](https://github.com/ros/ros_comm/blob/melodic-devel/utilities/message_filters/include/message_filters/synchronizer.h#L358).

According to [this StackOverflow post](https://stackoverflow.com/questions/44367859/using-stdarray-and-using-array-as-name) having `this->add<i>(evt)` is ill-formed. The solution is to add `template` so that the function becomes `this->template add<i>(evt)` this seems to also have been a problem before as the same `template` is present on [line 342](https://github.com/ros/ros_comm/blob/melodic-devel/utilities/message_filters/include/message_filters/synchronizer.h#L342).

I'll open an issue on `ros_comm` instead of here. I'll leave this issue open for now, but this can probably be closed.

mandraga, 2022-03-09:
Same problem using Yocto, I had to patch the file synchronizer.h plus adding a patchdir argument to my src_uri.
SRC_URI += "file://0001-FIX-template-c-error.patch;patchdir=${WORKDIR}/${ROS_SP}"