ros-perception/image_common #93: [ros2] Remapping arguments passed to node constructor are ignored

chapulina, 2018-09-13:
If I pass a remapping argument to a node's constructor like this: 
    
    node_ = rclcpp::Node::make_shared(..., "old_topic:=new_topic", ...);

And then create a publisher like this:

    auto pub = image_transport::create_publisher(node_, "old_topic");

I'd expect the publisher to publish on `new_topic`, but it publishes on `old_topic`.

I've written a failing test on https://github.com/chapulina/image_common/commit/403b8c2013523a3a87bb40783c5a91e4c5b22a79

mjcarroll, 2018-09-13:
Thanks, I would expect that the remapping would work, I'll take a look.

chapulina, 2018-09-14:
Once this is addressed, it would also be nice to add a function like `rclcpp::Publisher`'s which returns the fully qualified topic instead of just the base topic:

https://github.com/ros2/rclcpp/blob/86cc8fdb3ab06f496c718ca97998c1301b3af7d2/rclcpp/include/rclcpp/publisher.hpp#L74-L78

Right now, `image_transport::Publisher` only has the `getTopic` function, which returns the base topic.

https://github.com/ros-perception/image_common/blob/14d206622ad49e14eb412ccde04f77028c8f4a19/image_transport/include/image_transport/publisher.h#L88-L91

closed 2018-10-23 by mjcarroll