ros-perception/image_common #96: ROS2 image_transport raw topic rename issue

yechun1, 2018-10-11:
While I tried to use image_transport to publish camera raw image (/camera/color/image_raw) on ROS2, I found the topic has been renamed with /raw behind the basic topic( as /camera/color/image_raw/raw), this is not the same behavior compared with ROS1.

Is this ROS2 behavior changes? or image_transport porting issue? or am I use the wrong APIs? thanks for help.

My code:

```
image_transport::Publisher pub
image_transport::ImageTransport image_transport(node_);
pub = image_transport.advertise("camera/color/image_raw", 1);
```

Actually: Topic with image_transport:
> **/camera/color/image_raw/raw**
> /camera/color/image_raw/compressed
> /camera/color/image_raw/compressedDepth

Expect: What expected is keep the same topic name with basic topic name:
> **/camera/color/image_raw**
> /camera/color/image_raw/compressed
> /camera/color/image_raw/compressedDepth

Compare with ROS 1 image_transport:
> **/camera/color/image_raw**  (no /raw append)
> /camera/color/image_raw/compressed
> /camera/color/image_raw/compressed/parameter_descriptions
> /camera/color/image_raw/compressed/parameter_updates
> /camera/color/image_raw/compressedDepth
> /camera/color/image_raw/compressedDepth/parameter_descriptions
> /camera/color/image_raw/compressedDepth/parameter_updates

mjcarroll, 2018-10-16:
Ah, not entirely a duplicate.  I'm looking into it.

mjcarroll, 2018-10-16:
Yes, this was an oversight when porting, #97 addresses it.

closed 2018-10-23 by mjcarroll

yechun1, 2018-10-25:
@mjcarroll thanks. The issue fixed by the remapping patch #97