ros-perception/image_common #107: image_transport build error with ros2

sfutlan, 2018-12-12:
ubuntu 18.04
I was successfully Installed ROS2 via Debian Packages by the link
https://index.ros.org/doc/ros2/Linux-Install-Debians/

I was trying to install realsense sdk 2.0 by the link
https://github.com/intel/ros2_intel_realsense

when I tried to colcon build image_transport package,I got errors, thanks for any helps.

$ cd ~/ros2_ws/src
$ git clone https://github.com/ros-perception/image_common.git
$ git checkout ros2
$ cd ~/ros2_ws
$ colcon build --base-paths src/image_common/image_transport

u@u-OptiPlex-9020:~/ros2_ws$ colcon build --base-paths src/image_common/image_transport
Starting >>> image_transport
--- stderr: image_transport                               
In file included from /opt/ros/bouncy/include/rclcpp/service.hpp:33:0,
                 from /opt/ros/bouncy/include/rclcpp/callback_group.hpp:24,
                 from /opt/ros/bouncy/include/rclcpp/node.hpp:36,
                 from /home/u/ros2_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h:38,
                 from /home/u/ros2_ws/src/image_common/image_transport/include/image_transport/raw_publisher.h:39,
                 from /home/u/ros2_ws/src/image_common/image_transport/src/manifest.cpp:36:
/home/u/ros2_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h: In member function ‘virtual void image_transport::SimplePublisherPlugin<M>::publish(const Image&) const’:
/opt/ros/bouncy/include/rclcpp/logging.hpp:362:83: error: type/value mismatch at argument 1 in template parameter list for ‘template<class, class> struct std::is_same’
     ::std::is_same<std::remove_reference<decltype(logger)>::type, ::rclcpp::Logger>::value, \
                                                                                   ^
/home/u/ros2_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h:91:7: note: in expansion of macro ‘RCLCPP_ERROR’
       RCLCPP_ERROR(simple_impl_->logger_,
       ^~~~~~~~~~~~
/opt/ros/bouncy/include/rclcpp/logging.hpp:362:83: note:   expected a type, got ‘std::remove_reference<decltype (((const image_transport::SimplePublisherPlugin<M>*)this)->image_transport::SimplePublisherPlugin<M>::simple_impl_->logger_)>::type’
     ::std::is_same<std::remove_reference<decltype(logger)>::type, ::rclcpp::Logger>::value, \
                                                                                   ^
/home/u/ros2_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h:91:7: note: in expansion of macro ‘RCLCPP_ERROR’
       RCLCPP_ERROR(simple_impl_->logger_,
       ^~~~~~~~~~~~
/home/u/ros2_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h: In member function ‘virtual void image_transport::SimplePublisherPlugin<M>::advertiseImpl(rclcpp::Node*, const string&, rmw_qos_profile_t)’:
/opt/ros/bouncy/include/rclcpp/logging.hpp:68:83: error: type/value mismatch at argument 1 in template parameter list for ‘template<class, class> struct std::is_same’
     ::std::is_same<std::remove_reference<decltype(logger)>::type, ::rclcpp::Logger>::value, \
                                                                                   ^
/home/u/ros2_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h:112:5: note: in expansion of macro ‘RCLCPP_DEBUG’
     RCLCPP_DEBUG(simple_impl_->logger_, "getTopicToAdvertise: %s", transport_topic.c_str());
     ^~~~~~~~~~~~
/opt/ros/bouncy/include/rclcpp/logging.hpp:68:83: note:   expected a type, got ‘std::remove_reference<decltype (((image_transport::SimplePublisherPlugin<M>*)this)->image_transport::SimplePublisherPlugin<M>::simple_impl_->logger_)>::type’
     ::std::is_same<std::remove_reference<decltype(logger)>::type, ::rclcpp::Logger>::value, \
                                                                                   ^
/home/u/ros2_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h:112:5: note: in expansion of macro ‘RCLCPP_DEBUG’
     RCLCPP_DEBUG(simple_impl_->logger_, "getTopicToAdvertise: %s", transport_topic.c_str());
     ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/image_transport_plugins.dir/src/manifest.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_transport_plugins.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

Failed   <<< image_transport	[ Exited with code 2 ]

Summary: 0 packages finished [12.6s]
  1 package failed: image_transport
  1 package had stderr output: image_transport

chapulina, 2018-12-12:
The latest `ros2` branch works with Crystal, but you're trying to compile it against Bouncy.

mjcarroll, 2019-01-17:
@sfutlan without any additional information, I'm going to close this ticket.  This should be resolved by either using the Crystal release or current master.

closed 2019-01-17 by mjcarroll