ros-perception/image_common #111: Enable "lazy" publishers and subscribers

mjcarroll, 2019-01-17:
## Feature request

(From ros2/ros2#531)

#### Feature description

Using the `count_matched` capabilities of RMW, allow publishers to enable/disable work based on the number of subscribers present.

#### Implementation considerations

Can utilize: ros2/rmw#155, which still needs to be exposed at the `rclcpp` level.

StefanFabian, 2021-01-19:
Is there any update on this feature?

mikeferguson, 2023-05-17:
I'm just starting to work on this since we have matched subscribers on Iron - I think we need a minor API tweak (new parameters to pass in PublisherOptions, etc for subscriber).

@clalancette it looks like you cut the last release here - wondering where this package falls in terms of API change cutoff with regards to Iron.

mikeferguson, 2023-05-17:
@clalancette is that true even if we add another parameter BUT have a default so that all code above it will compile and have unchanged behavior? (which is what I have in #272).

The lower level stuff barely made it in before the cut off so I didn't have time to do the higher level things like this - but it would kinda stink to have wait another year before we can actually get lazy subscribers in image_proc/depth_image_proc

clalancette, 2023-05-17:
> @clalancette is that true even if we add another parameter BUT have a default so that all code above it will compile and have unchanged behavior? (which is what I have in #272).

Unfortunately, yes.  That changes ABI.

mikeferguson, 2023-05-17:
@clalancette As a note - we should probably update how we describe the freeze in the future - https://discourse.ros.org/t/upcoming-feature-freeze-for-ros-2-iron-irwini-on-17th-april-2023/30812 calls out that only the base/core packages are frozen - but the list on https://www.ros.org/reps/rep-2001.html#id26 shows image_common is part of the "perception" which extends base - to me, that read as though we hadn't frozen image_common april 17th

mikeferguson, 2024-02-03:
#272 was merged some time ago, so this will be part of the Jazzy release

closed 2024-02-03 by mikeferguson