ros-perception/image_common #121: image_transport build error with ros2

congliu0913, 2019-02-14:
Ubuntu 18.04

I've already installed ROS2 crystal via source code build (master branch), but build image_transport fail, below are fail log.

image_transport                                                                                                                                                                              
In file included from /usr/include/x86_64-linux-gnu/c++/7/bits/c++allocator.h:33:0,
                 from /usr/include/c++/7/bits/allocator.h:46,
                 from /usr/include/c++/7/string:41,
                 from /usr/include/c++/7/bits/locale_classes.h:40,
                 from /usr/include/c++/7/bits/ios_base.h:41,
                 from /usr/include/c++/7/ios:42,
                 from /usr/include/c++/7/ostream:38,
                 from /home/intel/rap_framework/out_Release/amr/ros2/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:55,
                 from /home/intel/rap_framework/src/amr/ros2/modules/ros-perception/image_common/image_transport/test/test_remapping.cpp:1:
/usr/include/c++/7/ext/new_allocator.h: In instantiation of ‘void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = rclcpp::Node; _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}; _Tp = rclcpp::Node]’:
/usr/include/c++/7/bits/alloc_traits.h:475:4:   required from ‘static void std::allocator_traits<std::allocator<_CharT> >::construct(std::allocator_traits<std::allocator<_CharT> >::allocator_type&, _Up*, _Args&& ...) [with _Up = rclcpp::Node; _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}; _Tp = rclcpp::Node; std::allocator_traits<std::allocator<_CharT> >::allocator_type = std::allocator<rclcpp::Node>]’
/usr/include/c++/7/bits/shared_ptr_base.h:526:39:   required from ‘std::_Sp_counted_ptr_inplace<_Tp, _Alloc, _Lp>::_Sp_counted_ptr_inplace(_Alloc, _Args&& ...) [with _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}; _Tp = rclcpp::Node; _Alloc = std::allocator<rclcpp::Node>; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2]’
/usr/include/c++/7/bits/shared_ptr_base.h:637:4:   required from ‘std::__shared_count<_Lp>::__shared_count(std::_Sp_make_shared_tag, _Tp*, const _Alloc&, _Args&& ...) [with _Tp = rclcpp::Node; _Alloc = std::allocator<rclcpp::Node>; _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2]’
/usr/include/c++/7/bits/shared_ptr_base.h:1295:35:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_make_shared_tag, const _Alloc&, _Args&& ...) [with _Alloc = std::allocator<rclcpp::Node>; _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}; _Tp = rclcpp::Node; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2]’
/usr/include/c++/7/bits/shared_ptr.h:344:64:   required from ‘std::shared_ptr<_Tp>::shared_ptr(std::_Sp_make_shared_tag, const _Alloc&, _Args&& ...) [with _Alloc = std::allocator<rclcpp::Node>; _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}; _Tp = rclcpp::Node]’
/usr/include/c++/7/bits/shared_ptr.h:690:14:   required from ‘std::shared_ptr<_Tp> std::allocate_shared(const _Alloc&, _Args&& ...) [with _Tp = rclcpp::Node; _Alloc = std::allocator<rclcpp::Node>; _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}]’
/usr/include/c++/7/bits/shared_ptr.h:706:39:   required from ‘std::shared_ptr<_Tp> std::make_shared(_Args&& ...) [with _Tp = rclcpp::Node; _Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}]’
/home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/node.hpp:71:3:   required from ‘static std::shared_ptr<rclcpp::Node> rclcpp::Node::make_shared(Args&& ...) [with Args = {const char (&)[11], const char (&)[10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >&}]’
/home/intel/rap_framework/src/amr/ros2/modules/ros-perception/image_common/image_transport/test/test_remapping.cpp:32:5:   required from here
/usr/include/c++/7/ext/new_allocator.h:136:4: error: no matching function for call to ‘rclcpp::Node::Node(const char [11], const char [10], std::shared_ptr<rclcpp::contexts::default_context::DefaultContext>&, std::vector<std::__cxx11::basic_string<char> >&, std::vector<rclcpp::Parameter>&)’
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp:28:0,
                 from /home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/executors.hpp:22,
                 from /home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/rclcpp.hpp:144,
                 from /home/intel/rap_framework/src/amr/ros2/modules/ros-perception/image_common/image_transport/test/test_remapping.cpp:7:
/home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/node.hpp:90:12: note: candidate: rclcpp::Node::Node(const string&, const string&, const rclcpp::NodeOptions&)
   explicit Node(
            ^~~~
/home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/node.hpp:90:12: note:   candidate expects 3 arguments, 5 provided
/home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/node.hpp:79:12: note: candidate: rclcpp::Node::Node(const string&, const rclcpp::NodeOptions&)
   explicit Node(
            ^~~~
/home/intel/rap_framework/out_Release/amr/ros2/install/rclcpp/include/rclcpp/node.hpp:79:12: note:   candidate expects 2 arguments, 5 provided
make[2]: *** [CMakeFiles/image_transport-remapping.dir/test/test_remapping.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_transport-remapping.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

congliu0913, 2019-02-14:
it's seems ROS2 RCLCPP updated, https://github.com/ros2/rclcpp/commit/10d7b7c72bc243e4e6da0546f9d17e39460df55c.

yechun1, 2019-02-15:
I meet the same error, test_remapping.cpp build fails as rclcpp Node changed. https://github.com/ros2/rclcpp/commit/0f9098e9b65169c9c4902a3b504733144f4bcab2

kungfrank, 2019-03-29:
Anyone figure out how to solve this problem? I meet the same error too.

nightduck, 2020-05-21:
Bump. Same issue. Could this be a toolchain thing? When I use clang6, I get "no matching constructor for initialization of 'image_transport::ImageTransport'", even though I'm using the exact code recommended [here](https://github.com/ros-perception/image_common/wiki/ROS2-Migration).

Build log attached:
```
robotos@jetson-agx:~/ros2_ws$ colcon build --packages-select opencv_demos
Starting >>> opencv_demos
--- stderr: opencv_demos                               
/home/robotos/ros2_ws/src/opencv_demos/src/image_converter.cpp:22:9: error: static_assert failed due to requirement '::std::is_same<typename std::remove_reference<typename std::remove_cv<decltype("cv_bridge exception: %s")>::type>::type, typename ::rclcpp::Logger>::value' "First argument to logging macros must be an rclcpp::Logger"
        RCLCPP_ERROR("cv_bridge exception: %s", e.what());
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/dashing/include/rclcpp/logging.hpp:490:5: note: expanded from macro 'RCLCPP_ERROR'
    static_assert( \
    ^
/home/robotos/ros2_ws/src/opencv_demos/src/image_converter.cpp:22:9: error: member reference base type 'const char [24]' is not a structure or union
        RCLCPP_ERROR("cv_bridge exception: %s", e.what());
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/dashing/include/rclcpp/logging.hpp:495:13: note: expanded from macro 'RCLCPP_ERROR'
      logger.get_name(), \
      ~~~~~~^~~~~~~~~
/opt/ros/dashing/include/rcutils/logging_macros.h:996:63: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    RCUTILS_LOG_CONDITION_EMPTY, RCUTILS_LOG_CONDITION_EMPTY, name, \
                                                              ^~~~
/opt/ros/dashing/include/rcutils/logging_macros.h:70:47: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
    if (rcutils_logging_logger_is_enabled_for(name, severity)) { \
                                              ^~~~
/home/robotos/ros2_ws/src/opencv_demos/src/image_converter.cpp:22:9: error: member reference base type 'const char [24]' is not a structure or union
        RCLCPP_ERROR("cv_bridge exception: %s", e.what());
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/dashing/include/rclcpp/logging.hpp:495:13: note: expanded from macro 'RCLCPP_ERROR'
      logger.get_name(), \
      ~~~~~~^~~~~~~~~
/opt/ros/dashing/include/rcutils/logging_macros.h:996:63: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    RCUTILS_LOG_CONDITION_EMPTY, RCUTILS_LOG_CONDITION_EMPTY, name, \
                                                              ^~~~
/opt/ros/dashing/include/rcutils/logging_macros.h:72:58: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                         ^~~~
/home/robotos/ros2_ws/src/opencv_demos/src/image_converter.cpp:43:37: error: no matching constructor for initialization of 'image_transport::ImageTransport'
    image_transport::ImageTransport it(node_.get());
                                    ^  ~~~~~~~~~~~
/opt/ros/dashing/include/image_transport/image_transport.h:108:8: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'std::__shared_ptr<rclcpp::Node, __gnu_cxx::_S_atomic>::element_type *' (aka 'rclcpp::Node *') to 'const image_transport::ImageTransport' for 1st argument
class  ImageTransport
       ^
/opt/ros/dashing/include/image_transport/image_transport.h:116:12: note: candidate constructor not viable: no known conversion from 'std::__shared_ptr<rclcpp::Node, __gnu_cxx::_S_atomic>::element_type *' (aka 'rclcpp::Node *') to 'rclcpp::Node::SharedPtr' (aka 'shared_ptr<rclcpp::Node>') for 1st argument
  explicit ImageTransport(rclcpp::Node::SharedPtr node);
           ^
4 errors generated.
make[2]: *** [CMakeFiles/image_converter.dir/src/image_converter.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_converter.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< opencv_demos	[ Exited with code 2 ]

Summary: 0 packages finished [14.1s]
  1 package failed: opencv_demos
  1 package had stderr output: opencv_demos

```

nightduck, 2020-05-22:
The workaround is to avoid instantiating an ImageTransport object, and use the static create_publisher and create_subscription methods. See [gist](https://gist.github.com/nightduck/a07c185faad82aeaacbfa87298d035c0) for example

    rmw_qos_profile_t custom_qos = rmw_qos_profile_default;
    pub_ = image_transport::create_publisher(this, "out_image_base_topic", custom_qos);
    sub_ = image_transport::create_subscription(this, "in_image_base_topic",
            std::bind(&ImageConverter::imageCallback, this, std::placeholders::_1), "raw", custom_qos);