ros-perception/image_common #122: Segfault in the subscriber example

francofusco, 2019-03-05:
While following the tutorial [Writing a Simple Image Subscriber (C++)](http://wiki.ros.org/image_transport/Tutorials/SubscribingToImages) I got the following error:
`Attempt to unlock mutex that was not locked`
After reading this [issue in image_pipeline](https://github.com/ros-perception/image_pipeline/issues/201#issuecomment-400271494), I removed the line `cv::startWindowThread()`, and now it works fine. I do not know if this problem affects many other users, but it might be worth referencing this potential issue in the tutorial!

chyohoo, 2020-02-26:
I got the same problem, but removing the line `cv::startWindowThread()` didn't work for me.

chyohoo, 2020-02-26:
That works for me too.

SuYouge, 2020-05-30:
Thank you that works well.