ros-perception/image_common #139: [ros2] macOS Build failure due to future version requirements

zmk5, 2019-10-20:
Hi,

I'm trying to build the `image_common` package for macOS and I keep running into the following error output:

```bash
zmk5-MBP:ragesh_ws zmk5$ colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> camera_calibration_parsers
Starting >>> image_geometry
Starting >>> image_transport
Finished <<< image_geometry [1.51s]
Finished <<< cv_bridge [1.76s]
Starting >>> opencv_tests
Starting >>> vision_opencv
--- stderr: camera_calibration_parsers
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:13: error: 'path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:739:24: note: 'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:32: error: 'path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
                               ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:739:24: note: 'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:22: error: 'path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
                     ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:773:3: note: 'path' has been explicitly marked unavailable here
  path(const _Source& __src, format = format::auto_format) {
  ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:22: error: '~path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
                     ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:791:3: note: '~path' has been explicitly marked unavailable here
  ~path() = default;
  ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:48: error: 'parent_path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
                                               ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:1064:34: note: 'parent_path' has been explicitly marked unavailable here
  _LIBCPP_INLINE_VISIBILITY path parent_path() const {
                                 ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:48: error: '~path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
                                               ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:791:3: note: '~path' has been explicitly marked unavailable here
  ~path() = default;
  ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:18: error: 'path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
                 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:765:29: note: 'path' has been explicitly marked unavailable here
  _LIBCPP_INLINE_VISIBILITY path(path&& __p) noexcept
                            ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:306:18: error: '~path' is unavailable: introduced in macOS 10.15
  impl::fs::path dir(impl::fs::path(file_name).parent_path());
                 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:791:3: note: '~path' has been explicitly marked unavailable here
  ~path() = default;
  ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:307:12: error: 'empty' is unavailable: introduced in macOS 10.15
  if (!dir.empty() && !impl::fs::exists(dir) &&
           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:1077:3: note: 'empty' has been explicitly marked unavailable here
  empty() const noexcept {
  ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:307:34: error: 'exists' is unavailable: introduced in macOS 10.15
[... 672 lines cut ...]
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/pluginlib/./class_loader_imp.hpp:628:47: error: 'operator/' is unavailable: introduced in macOS 10.15
      std::string package_file_path = (parent / "package.xml").string();
                                              ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:1175:41: note: 'operator/' has been explicitly marked unavailable here
  friend _LIBCPP_INLINE_VISIBILITY path operator/(const path& __lhs,
                                        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
make[2]: *** [CMakeFiles/image_transport.dir/src/subscriber.cpp.o] Error 1
20 errors generated.
make[2]: *** [CMakeFiles/image_transport.dir/src/publisher.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_transport.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< image_transport	[ Exited with code 2 ]
Aborted  <<< opencv_tests
Aborted  <<< vision_opencv

Summary: 2 packages finished [2.29s]
  2 packages failed: camera_calibration_parsers image_transport
  2 packages aborted: opencv_tests vision_opencv
  2 packages had stderr output: camera_calibration_parsers image_transport
  1 package not processed
```

The error consistantly states that certain methods and functions are unavailable due to version macOS version requirements (e.g. `error: '~path' is unavailable: introduced in macOS 10.15`). Is there a way to get around this issue?

zmk5, 2020-01-28:
Hey, I think I may have figured it out. The issue seems to be that XCode has turned on the C++17 features but not the C++17 standard library in versions less than macOS 10.15. Therefore to be able to compile in Mojave and older, you will need to add the `-mmacosx-version-min=10.15` flag in your `CMakeLists.txt` file as such:

```make
if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic -mmacosx-version-min=10.15)
endif()
```

wolfv, 2021-02-22:
Hi @zmk5 I think it would be better to fix this in pluginlib (https://github.com/ros/pluginlib/blob/9fecba30c6e388024bc14137b53694b4e555696f/pluginlib/include/pluginlib/impl/filesystem_helper.hpp#L39-L49 ).

One could add a check for the OSX deployment target ... 

There are a bunch of polyfill filesystem libraries available as well that could be used.

wolfv, 2021-02-22:
@zmk5 I opened an issue here: https://github.com/ros/pluginlib/issues/215 maybe we can get that sorted out in the future. 

We got quite some ROS packages for Foxy on the robostack channel now, but still missing RViz 2 (because there were issues compiling RViz. If you have ideas, please help! :)

(Failing RViz build here: https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=1483&view=logs&j=72f35559-b34c-5aff-f26a-5f9b23141180&t=5a86ea88-b89b-5155-3f25-817bbf591b05 (sorry for hijacking this thread).