ros-perception/image_common #152: [ros2] camera_calibration_parsers package fails to build

zmk5, 2020-01-28:
Hi,

I've been trying to build the `image_common`, but I keep running into a build failure on the macOS ROS 2 Dashing Diademata - Patch Release 4 binary version. I believe it's because the version of `rcpputils` is older than the one used for this package:

```bash
zmk5-laptop:some_ws zmk5$ colcon build --symlink-install
Starting >>> camera_calibration_parsers
Starting >>> cv_bridge
Starting >>> image_transport
Starting >>> image_geometry
Starting >>> maps_hough
--- stderr: camera_calibration_parsers
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:307:12: error: no member named 'empty' in 'rcpputils::fs::path'
  if (!dir.empty() && !rcpputils::fs::exists(dir) &&
       ~~~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:308:21: error: no member named 'create_directories' in namespace 'rcpputils::fs'
    !rcpputils::fs::create_directories(dir))
     ~~~~~~~~~~~~~~~^
2 errors generated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_ini.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:171:12: error: no member named 'empty' in 'rcpputils::fs::path'
  if (!dir.empty() && !rcpputils::fs::exists(dir) &&
       ~~~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:172:21: error: no member named 'create_directories' in namespace 'rcpputils::fs'
    !rcpputils::fs::create_directories(dir))
     ~~~~~~~~~~~~~~~^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:54:9: error: no member named 'extension' in 'rcpputils::fs::path'
  if (p.extension().string() == ".ini") {
      ~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:56:16: error: no member named 'extension' in 'rcpputils::fs::path'
  } else if (p.extension().string() == ".yml" || p.extension().string() == ".yaml") {
             ~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:56:52: error: no member named 'extension' in 'rcpputils::fs::path'
  } else if (p.extension().string() == ".yml" || p.extension().string() == ".yaml") {
                                                 ~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:62:9: error: no member named 'extension' in 'rcpputils::fs::path'
      p.extension().string().c_str());
      ~ ^
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rclcpp/logging.hpp:497:35: note: expanded from macro 'RCLCPP_ERROR'
        RCLCPP_ALL_BUT_FIRST_ARGS(__VA_ARGS__,"")); \
                                  ^~~~~~~~~~~
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rclcpp/logging.hpp:35:43: note: expanded from macro 'RCLCPP_ALL_BUT_FIRST_ARGS'
#define RCLCPP_ALL_BUT_FIRST_ARGS(N, ...) __VA_ARGS__
                                          ^~~~~~~~~~~
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rcutils/logging_macros.h:997:5: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    __VA_ARGS__)
    ^~~~~~~~~~~
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:73:9: error: no member named 'extension' in 'rcpputils::fs::path'
  if (p.extension().string() == ".ini") {
      ~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:75:16: error: no member named 'extension' in 'rcpputils::fs::path'
  } else if (p.extension().string() == ".yml" || p.extension().string() == ".yaml") {
             ~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:75:52: error: no member named 'extension' in 'rcpputils::fs::path'
  } else if (p.extension().string() == ".yml" || p.extension().string() == ".yaml") {
                                                 ~ ^
/Users/zmk5/ROS2/ragesh_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:81:9: error: no member named 'extension' in 'rcpputils::fs::path'
      p.extension().string().c_str());
      ~ ^
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rclcpp/logging.hpp:497:35: note: expanded from macro 'RCLCPP_ERROR'
        RCLCPP_ALL_BUT_FIRST_ARGS(__VA_ARGS__,"")); \
                                  ^~~~~~~~~~~
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rclcpp/logging.hpp:35:43: note: expanded from macro 'RCLCPP_ALL_BUT_FIRST_ARGS'
#define RCLCPP_ALL_BUT_FIRST_ARGS(N, ...) __VA_ARGS__
                                          ^~~~~~~~~~~
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rcutils/logging_macros.h:997:5: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    __VA_ARGS__)
    ^~~~~~~~~~~
/Users/zmk5/ROS2/ros2_install/ros2-osx/include/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
8 errors generated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse.cpp.o] Error 1
2 errors generated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_yml.cpp.o] Error 1
make[1]: *** [CMakeFiles/camera_calibration_parsers.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< camera_calibration_parsers	[ Exited with code 2 ]
Aborted  <<< maps_hough
Aborted  <<< image_geometry
Aborted  <<< image_transport
Aborted  <<< cv_bridge

Summary: 0 packages finished [3.37s]
  1 package failed: camera_calibration_parsers
  4 packages aborted: cv_bridge image_geometry image_transport maps_hough
  1 package had stderr output: camera_calibration_parsers
  4 packages not processed

```

ErikOrjehag, 2020-02-11:
I have the same problem

Luca1978, 2020-07-29:
Same problem for me....any solution available?

Luca1978, 2020-07-29:
I solved the problem manually substituiting rcpputils with the last code available on the rcpp repo at

https://github.com/ros2/rcpputils