ros-perception/image_common #154: Release image_common to ROS Noetic?

sloretz, 2020-03-10:
@vrabaud @jack-oquin would you mind to releasing `image_common` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 168 repos need it.

I'm not sure how much work it will need. At a minimum it needs #153. There also seem to be a few PRs with Python 3 fixes; which will be required. This wiki page has some useful info for transitioning a package to Python 3: http://wiki.ros.org/UsingPython3/

seanyen, 2020-03-26:
This pull request #155 is also helpful for Python3 support. As @sloretz mentioned, this is a package on the critical path of the ROS dependency graph and hope we can move forward and unblock them.

sloretz, 2020-04-13:
Released in https://github.com/ros/rosdistro/pull/24343 , Thanks @mjcarroll !

closed 2020-04-13 by sloretz