ros-perception/image_common #156: ROS2 ImageTransport subscribes with default QOS settings

mkhansenbot, 2020-03-31:
https://github.com/ros-perception/image_common/blob/e3f482846435eb65e299e9e5955788dc5bcf73c6/image_transport/src/image_transport.cpp#L159

It's fine this is the default but there should be an optional parameter to override it to something else, specifically the Sensor Default QOS (best effort, etc). Sensor Default is typically used with camera image data.

mkhansenbot, 2020-03-31:
I can submit a small PR to add the ability to override this default if needed, please reply either way.

(cc:@emersonknapp was the one who first found this issue)

emersonknapp, 2020-03-31:
My 2c - the internal default should be sensordata as a start. Then optional override is also useful

mkhansenbot, 2020-04-06:
Hoping to get a reply on this before Foxy release

ruffsl, 2020-04-17:
I think this could be solved by updating the underlying `message_filters` package to leverage the newer options API pattern directly: https://github.com/ros2/message_filters/issues/45

I didn't have the the chance to to try this, but you may want to take a look, @mkhansenbot .

mkhansenbot, 2020-05-14:
@mjcarroll @wjwwood - can either of you comment on this? I realize it's probably too late to fix for Foxy, but this seems like an important issue to fix

mkhansenbot, 2020-05-14:
@ruffsl - sorry I didn't reply earlier, but I don't quite understand how message_filters helps here

ruffsl, 2020-05-14:
It's used in the `image_transport` package, and would need to be updated so that a filter's subscribers also use the user selected QoS and message memory strategy when creating the image_transport:

https://github.com/ros-perception/image_common/blob/bd2a7f9c570f7ac7f24f1c284e53e87338482c58/image_transport/src/camera_subscriber.cpp#L51-L53

https://github.com/ros-perception/image_common/blob/bd2a7f9c570f7ac7f24f1c284e53e87338482c58/image_transport/src/camera_subscriber.cpp#L125-L128

wjwwood, 2020-05-14:
If it can be added in a non-API breaking way then it could still be put in foxy. If it can be done in a non-ABI breaking way it could be added to foxy after the release too.

I don't have time to pick it up, but if someone makes pull requests we'll try to get to them. @ me or @mjcarroll and we'll try to prioritize them if a non-API breaking, but ABI breaking change is required, as that's the most time sensitive version.

mkhansenbot, 2020-07-14:
Is there any plan to resolve this?

agutenkunst, 2021-02-07:
This issue caused me some pain and time until I figured out why no images are received. Would love to see a fix in foxy!

uxsapien, 2022-01-15:
Does this problem still persist. I'm using the Gazebo camera plugin in Galactic and would really like to adjust qos profile for the published image topic.

ihasdapie, 2022-06-15:
This should be resolved as of https://github.com/ros-perception/image_common/pull/246 for rolling. @mkhansenbot if this addresses the problem feel free to close this issue.