ros-perception/image_common #171: image_transport republish broken on Foxy?

KenYN, 2020-09-24:
I have the latest version installed, and I do the following:

```
$ ros2 topic list
/bar
/foo
/parameter_events
/rosout
```

OK, a couple of left-over topics (how do I flush everything out?) but otherwise OK

```
$ ros2 run image_transport republish compressed --ros-args --remap in:=/woo -- --ros-args --remap out:=/bazzle -- &
[1] 1661358
$ ros2 topic list
/bar
/bazzle
/foo
/in/compressed
/out/compressed
/parameter_events
/rosout
```

Oops, I expect to see:

```
$ ros2 run image_transport republish compressed --ros-args --remap in:=/woo -- --ros-args --remap out:=/bazzle -- &
[1] 1661358
$ ros2 topic list
/bar
/bazzle
/bazzle/compressed
/foo
/woo
/woo/compressed
/parameter_events
/rosout
```

It looks to me like the remapping has not been applied. I see the same problem with `image_transport::create_subscription()` and using `ComposableNode(..., remappings=[])`. `remappings` works fine, however, when using `rclcpp::Node::create_subscription()`, however.

KenYN, 2020-09-24:
It would seem to be related to this issue:

https://github.com/ros2/rclcpp/issues/985

GertVen, 2022-01-11:
Sorry to bring up an old question, but does anybody have a solution for the remapping, or is is the "out/" the correct way ? When creating an image transport I also get a bunch of "out/*" topics, for example:

`ros2 run image_transport republish raw --ros-args --remap in:=/a/image_raw --remap out:=/a/image_comp`
`ros2 run image_transport republish raw --ros-args --remap in:=/b/image_raw --remap out:=/b/image_comp`

Passes out:
`/a/image_raw
/a/image_comp
/b/image_raw
/b/image_comp
/out/compressed
/out/compressed_depth
/out/theora`

I am using ROS2 Galactic

If the data above is correct, what would be the best way to differentiate between two different input streams, for example a front and a rear camera.

Thanks!

Purg, 2022-03-02:
I think I have an "answer", or at least some sort of working usage. When you only pass the `ros2 run image_transport republish raw`, you are saying to "read in raw, republish all plugins" (thus all of compressed/depth/theora). It seems like the republish tool wants a `--remap` for every applicable `out`, e.g. `--remap out/compressed:=target/compressed --remap out/compressedDepth:=target/compressedDepth [etc.]`. If the original intention is to *not* have to be this verbose, it doesn't seem to work that way anymore.

In my particular use-case, I just want the compressed republish, so I've landed on:
```
ros2 run image_transport republish raw compressed --ros-args --remap in:=/target --remap out/compressed:=/target/compressed`
```
And wind up with the topics:
```
/target
/target/compressed
```

abourbou, 2023-05-17:
> I think I have an "answer", or at least some sort of working usage. When you only pass the `ros2 run image_transport republish raw`, you are saying to "read in raw, republish all plugins" (thus all [...]

Thank you the problem of topic is indeed resolved if you put the exact name_space you want to change depending on compression (in for raw else in/compressed for input and out for raw else out/compressed for output)

gitglob, 2026-02-13:
I have been struggling with a similar issue for the last 2 days. 
I wanted to convert a compressed image topic to an uncompressed one.
 
I finally realized that the correct command format was the following in my case:
```
ros2 run image_transport republish --ros-args -p in_transport:=compressed -p out_transport:=raw -r in/compressed:=camera_0/image_raw/compressed -r out:=camera_0/image_raw/uncom
pressed
``` 

I had to dive into the actual implementation to figure it out:
https://github.com/ros-perception/image_common/blob/rolling/image_transport/src/republish.cpp

It also helps a lot to inspect the node info to understand if your remappings are correct.