ros-perception/image_common #184: [ROS2 Foxy] `image_transport_plugins` broken

Myzhar, 2021-03-24:
There are two problems with `image_transport_plugins` that were not present in ROS2 Eloquent and they are both related to compressed image publisher.

The first time `create_camera_publisher` is called this error is thrown:
`[ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' could not be set: Not supported
`
When you call `create_camera_publisher` to instantiate other publishers:
`[ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
`
It seems that there is something wrong with the node namespace and the parameters namespace...

You can read more here:
https://github.com/stereolabs/zed-ros2-wrapper/issues/31

laszloturanyi, 2021-04-23:
This has been merged.
https://github.com/ros-perception/image_transport_plugins/pull/58
I tried to compile image_transport_plugins from source and it worked fine.

Void12987, 2022-02-10:
I can confirm ZED2 starts on Foxy Docker (specifically `sudo docker run --gpus all -it --privileged dustynv/ros:foxy-slam-l4t-r32.6.1`) but image_transport won't load so the image/depth data can't be echoed.

ruffsl, 2022-02-16:
> `dustynv/ros:foxy-slam-l4t-r32.6.1`

> [LAST PUSHED 6 months ago by dustynv](https://hub.docker.com/layers/dustynv/ros/foxy-slam-l4t-r32.6.1/images/sha256-c0d61d653c464c7d08309fd7af57b13ad68c4622da463dd97b8bb2938ef7522e?context=explore)

Has this image been updated with respect to the latest foxy sync?

closed 2025-10-08 by ahcorde