ros-perception/image_common #198: [camera_info_manager] Segmentation fault

MatthijsBurgh, 2021-07-09:
On a regular bases I get a segmentation fault in the camera_info_manager, in the `isCalibrated` function.

It is related to the `scoped_lock` in that function, https://github.com/ros-perception/image_common/blob/3eed8b5cf542d356051094535916ca180b2d3e6c/camera_info_manager/src/camera_info_manager.cpp#L187
I have tried with a with a `boost::lock_guard`, but also a `std::mutex` in combination with a `std::lock_guard`. I keep getting seg faults on that line.

The `camera_info_manager` is used as part of the `openni2_driver` which is run as a nodelet.

To me it isn't logical the lock should result in a segfault. I have run many gdb sessions, but I couldn't find anything else that catched my eye.

MatthijsBurgh, 2021-07-13:
@vrabaud @jack-oquin @mjcarroll You got any idea about this?

devrite, 2021-07-31:
One reason might be you using camera_info_manager  when it is already destroyed or not fully constructed yet (some race). In which case the mutex can be invalid to use.

Another cause:

I had boost issues before. In my case another proprietary API exported boost symbols of an incompatible version and started using those symbols. In this case you may inspect whether or not you end up in a function symbols of another DSO not part of your lib.