ros-perception/image_common #214: #211 breaks foxy builds

nkalupahana, 2021-11-13:
ROS2 Foxy currently builds off of the `ros2` branch, since it doesn't have its own distro branch. The `ros2` branch was recently updated with breaking changes to a header name (from #211) -- but this header hasn't been updated in Foxy, so this package is no longer buildable on Foxy.

Can we get a foxy branch on this repo without this header name change so that way foxy builds can progress? (Any other fix is also fine, this is just the one that makes the most sense to me.)

Pinging @aprotyas @clalancette @mjcarroll since they were involved with #211

aprotyas, 2021-11-14:
Thanks for the report @nkalupahana and apologies for any inconvenience.

Branching off from the `ros2` branch and reverting https://github.com/ros-perception/image_common/commit/a90cedf3693c3d3dccd19ae6425babb83189cf9b should do the trick, but I don't have push access to this repo so I'll leave that to the maintainers.

clalancette, 2021-11-15:
Good call.  I've made a `foxy` branch based off of the latest released commit into Foxy (2.3.0), and have opened a PR to rosdistro to switch to that branch for Foxy: https://github.com/ros/rosdistro/pull/31236

closed 2021-11-15 by clalancette