ros-perception/image_common #224: SimpleSubscriberPlugin shutdown() implementation

afrixs, 2022-02-03:
There is a [TODO](https://github.com/ros-perception/image_common/blob/9729de81f7dff6156f644d6152b200f687360f1f/image_transport/include/image_transport/simple_subscriber_plugin.hpp#L86) in SimpleSubscriberPlugin about waiting for ROS2 to support subscriber shutdown.

I believe ROS2 now supports subscriber shutdown - it's performed in subscriber's destructor.

I would propose to simply use `impl_.reset();` like it's done e.g. in [`message_filters` package](https://github.com/ros2/message_filters/blob/210fcb02a39bb2e674d6c4ad6e87526aa3927481/include/message_filters/subscriber.h#L319). I've tested it and it works.

gbiggs, 2022-02-04:
Could you please submit a PR that includes a test?

afrixs, 2022-03-11:
Reopening since calling `publish` on a disabled publisher caused process to crash (see #235)

closed 2022-03-25 by gbiggs