ros-perception/image_common #238: image_transport::CameraPublisher::getNumSubscribers

mhubii, 2022-03-27:
`image_transport::CameraPublisher::getNumSubscribers` does not return the correct number of subscribers in `foxy` distributed binaries. Building this repo from source mitigates this issue. Minimal example:

```cpp
#include <rclcpp/rclcpp.hpp>
#include <image_transport/image_transport.hpp>

void cb(const sensor_msgs::msg::Image::ConstSharedPtr) { return; }

int main(int argc, char** argv) {
    rclcpp::init(argc, argv);
    auto node = std::make_shared<rclcpp::Node>("tmp_node");

    std::string topic = "image";
    auto pub = image_transport::create_camera_publisher(node->shared_from_this().get(), topic);
    auto sub = image_transport::create_subscription(node->shared_from_this().get(), topic, cb, "raw");

    while (rclcpp::ok()) {
        RCLCPP_INFO(node->get_logger(), "N subscriptions: %d", pub.getNumSubscribers());
        rclcpp::spin_some(node);
        std::this_thread::sleep_for(std::chrono::seconds(1));
    }
    rclcpp::shutdown();
}
```

ie this will print `N subscriptions: 0`.

Full example: https://github.com/mhubii/image_transport_bug

mhubii, 2022-03-28:
downstream issues are for example ros-perception/image_pipeline#739

gbiggs, 2022-07-06:
So this bug is fixed in source but not in the released binaries? Having a new binary release would solve your problem?

mhubii, 2022-07-07:
I suppose

gbiggs, 2022-07-12:
I'll close this issue, then. If you encounter the bug again, please comment here and I'll reopen it and we can try to fix the problem.

mhubii, 2022-07-12:
I don't think closing without checking makes sense

gbiggs, 2022-07-12:
OK, I'll re-open it. But this is reported as working in source, and I'm doing a new binary release right now to get that version into the repositories. Please feel free to test and report once the next foxy sync happens if the bug is still present in the new binaries

mhubii, 2022-11-02:
hey @gbiggs , I checked and the error seems gone!

closed 2022-11-02 by mhubii