ros-perception/image_common #239: How to deactivate those topics which are not needed?

Shibaditya99, 2022-05-17:
I've successfully transfered the video stream one to another process with it. But in the ros topic list I can see there's several topics are activate like "compressed", "theora" etc. which I don't need. So how to deactivate those topics? I only need the raw data.
Thank you so much.

Petros626, 2022-05-19:
https://github.com/ros-perception/image_common/pull/60/commits/f4875d84eb5f587967d51ec0fd5242e6e1338995

wodtko, 2022-05-20:
I actually came across this the last few days, too.
It seems like there was the option to deactivate plugins by adding them to the parameter `disable_pub_plugins`.
However, with the current ros2 branch, I'm slightly confused about what happened with reading the parameter.

With #84 the line reading the parameter was commented out, but the blacklist remained. 
From how I understand the current code, the blacklist is always empty, no matter what value is set for the parameter.
This line (https://github.com/ros-perception/image_common/blob/ros2/image_transport/src/publisher.cpp#L115) should actually be used, shouldn't it?

closed 2022-06-24 by Shibaditya99

Ronoman, 2023-02-27:
Is there a reason this line of code couldn't be uncommented? I'm running humble and am looking to disable compressed image transport without needing to remove packages, as suggested by https://answers.ros.org/question/255970/how-to-deactivate-image_transport-plugins/.