ros-perception/image_common #240: [ROS2] error output when advertising publisher

wodtko, 2022-05-19:
The function `PublisherPlugin::advertiseImpl` was extended to support `PublisherOptions` with the last update.
If a child class does not override this function, an error message is generated and the `advertise` function without `PublisherOptions` is called instead.

However, the "default" `image_transport::advertise` function always creates `PublisherOptions` leading to error messages, even if the PublisherOptions are not required.
I would suggest using `std::optional<rclcpp::PublisherOptions>` in order to differentiate between options being required or not.
By this, the fitting `advertiseImpl` function can be called respectively.
Error messages are then only produced if the new option is explicitly required.

Additionally, the `SimplePublisherPlugin` seems to name the respective new function `advertiseImplWithOptions` instead of `advertiseImpl` which causes the error output for me in the first place.

wodtko, 2022-05-19:
PR available (https://github.com/ros-perception/image_common/pull/241)

wodtko, 2022-05-19:
If changing the functions' signatures is not an option, at least the `advertiseImplWithOptions` should be renamed.

wodtko, 2022-05-19:
by looking at the subscriber code, I found a similar structure.
the `SimplePublisherPlugin` is also calling the new function `[...]WithOptions`
Is there a specific reason which I am not yet aware of?

for the `RawPublisher` the base class call can simply be updated to call the new function.
For now, I can't see the reason for changing the way function overriding is used in contrast to without options.

gbiggs, 2022-07-26:
Relates to #249

gbiggs, 2022-07-27:
Fixed by #249.

closed 2022-07-27 by gbiggs