ros-perception/image_common #248: [Feature Request] Python bindings

Interpause, 2022-06-28:
I saw on <http://wiki.ros.org/image_transport> that Python support was on the roadmap, is there any issue tracking the progress?

gbiggs, 2022-06-28:
Unfortunately, I don't think there is. I'm not sure what the current state is.

Interpause, 2022-06-29:
I see... if I get desperate enough I will try learning pybind to see if I can help.

tfoldi, 2024-09-22:
This feature could actually be useful in my project. If no one is working on it, I am happy to implement it (add Python classes via pybind11 bindings).

ahcorde, 2024-09-23:
Hi @tfoldi, 

As a maintainer, I'm no aware of someone working on this feature. 

I implemented the Python bindings for point cloud transport https://github.com/ros-perception/point_cloud_transport just in case you need some inspiration.

Happy to review your PR

tfoldi, 2024-09-23:
sounds great, will start working on it. based on the repo you shared will use the same standards, like:

1. there will be a new top level folder called `image_common` 
2. keep a similar `pybind11.h` along with sources
3. test cases are not mandatory

Jack17432, 2024-10-30:
Would it be possible to have this updated for iron and humble?

ahcorde, 2025-09-03:
Python binding already merged https://github.com/ros-perception/image_common/tree/rolling/image_transport_py