ros-perception/image_common #279: Merge #276 broke downstream packages

jmachowinski, 2023-07-31:
Hi, 
the recent merge of 'Expose callback groups - backport to iron'
just broke donwstream packages (e.g. https://github.com/RobotWebTools/web_video_server) , as 
```c++
  Subscriber subscribe(
    const std::string & base_topic, uint32_t queue_size,
    void (* fp)(const ImageConstPtr &),
    const TransportHints * transport_hints = nullptr)
``` 
and 
```c++
  Subscriber subscribe(
    const std::string & base_topic, uint32_t queue_size,
    void (* fp)(const ImageConstPtr &),
    const TransportHints * transport_hints = nullptr,
    const rclcpp::SubscriptionOptions options = rclcpp::SubscriptionOptions())
``` 
are ambiguous.

adityapande-1995, 2023-08-02:
Opened a PR to revert the change here : https://github.com/ros-perception/image_common/pull/281 until I fix the issue.

jmachowinski, 2023-08-22:
Recent merges fixed the issue

closed 2023-08-22 by jmachowinski