ros-perception/image_common #341: Handling namespace for multiple CameraInfoManagers in ROS2 jazzy

PfeifferMicha, 2025-02-04:
Hi,

I'm trying to set up the driver for a stereo camera, so I need two CameraInfoManager instances.
The hope was that I could set up the classes in such a way that they advertise:
```
/stereo/node_name/left/set_camera_info and 
/stereo/node_name/right/set_camera_info
```
However, they always seem to end up advertising the exact same service name...?
```
/stereo/node_name/set_camera_info
```

Is there a workaround?

I saw that on the rolling branch, there is a "namespace" (ns) parameter that one can pass to the CameraInfoManager constructor, however it's missing on the jazzy branch. Is there a workaround? Or could this be backported?

sloretz, 2025-02-04:
The feature you're looking for was added in #324, and it was not backported because it breaks ABI. It does mention

> Up until Jazzy this could have been remedied by using a sub-node with its custom namespace, however since updating the code to work with lifecycle nodes this is not possible anymore.

If you're using normal nodes, you could [create a sub-node](https://github.com/ros2/rclcpp/blob/9dd2c91d7d99e872b2756791e82ddb62bc1ef5ec/rclcpp/include/rclcpp/node.hpp#L1573-L1613) and pass it to this constructor https://github.com/ros-perception/image_common/blob/34ad5dbd4ba9f45553b9b09887c4db39cdf03180/camera_info_manager/src/camera_info_manager.cpp#L80-L86

sloretz, 2025-02-05:
> I was expecting to have these services, which don't exist:
> ```
> /stereo/left_fullscale/set_camera_info
> /stereo/right_fullscale/set_camera_info
> ```

I would expect the same given that [the service uses the name `~/set_camera_info`](https://github.com/ros-perception/image_common/blob/34ad5dbd4ba9f45553b9b09887c4db39cdf03180/camera_info_manager/src/camera_info_manager.cpp#L109-L110) in Jazzy. This seems like a bug to me, but I'm not sure where yet. Maybe it's because [`rclcpp::create_service` takes a `NodeBaseInterface`, and that doesn't include sub-node info](https://github.com/ros2/rclcpp/blob/2cc43b327405f82d8192267714d509074ef2915f/rclcpp/include/rclcpp/create_service.hpp#L60)? (@wjwwood does that sound like a possible cause?)

> Is this related to the fact that I'm running all of this in a component? Does the component container somehow "swallow" the services?

I would not expect the component container to affect this. Components share the context object, but IIRC remapping names happens on Nodes, and components don't share Nodes. I'm very surprised by the name `/stereo/split_node_container/set_camera_info`.

I think next step to resolving this is to come up with a minimal reproducible example, maybe using just the `rclcpp` APIs used `CameraInfoManager`. @PfeifferMicha Is that something you'd be willing to take on?

PfeifferMicha, 2025-02-06:
Sure, [here's a minimal example](https://github.com/PfeifferMicha/test_cam_info_manager) using two sub-nodes and two CameraInfoManagers. I used jazzy to run this, but it'd be interesting to see if other distros are affected as well.

While the node is running, I see one `set_camera_info` service in the main node's namespace, even though clearly two are created (and they use two different sub-nodes):
```
/test_cam_info_manager/describe_parameters
/test_cam_info_manager/get_parameter_types
/test_cam_info_manager/get_parameters
/test_cam_info_manager/get_type_description
/test_cam_info_manager/list_parameters
/test_cam_info_manager/**set_camera_info**
/test_cam_info_manager/set_parameters
/test_cam_info_manager/set_parameters_atomically
```

I've also been searching for other repositories who use CameraInfoManager with stereo cameras, but I've not had any luck of finding a good, simple, working example. I thought this was the standard way of doing things?

PfeifferMicha, 2025-03-14:
A user named wjwwood explaines the underlying issue here: https://github.com/ros2/rclcpp/issues/2734#issuecomment-2691429428
They  have three suggestion of how to handle the problem, but I'm not familiar with the image_common package to really judge what's best.
Could someone chime in either here or on the upstream issue?