ros-perception/image_common #354: [ROS2] camera_info_manager_py uses wrong CameraInfo attributes names in ROS2

andrerusso02, 2025-04-11:
I encountered the following error when calling `camera_info_manager.loadCalibrationFile`.

```
AttributeError: 'CameraInfo' object has no attribute 'D'. Did you mean: 'd'?
```

I looked at the code and the issue is that it seems the package has not been properly ported to ROS2 ? For example [here](https://github.com/ros-perception/image_common/blob/jazzy/camera_info_manager_py/camera_info_manager/camera_info_manager.py#L473):

```
    ci = CameraInfo()
...
  ci.D = calib['distortion_coefficients']['data']
  ci.K = calib['camera_matrix']['data']
  ci.R = calib['rectification_matrix']['data']
  ci.P = calib['projection_matrix']['data']
```

It uses uppercase letters for the matrices attributes, which corresponds to the ROS1 CameraInfo message definition. But in ROS2, [they are lowercase](https://docs.ros2.org/latest/api/sensor_msgs/msg/CameraInfo.html).

I'm a bit confused. Am I dooing something wrong here? It really seems to me like an issue with the package, but that's such a big issue that I don't understand how it could have went unnoticed for such a long period of time.

xuyuan, 2025-06-03:
I just found the same issue, seems nobody really use the package?

Rick-v-E, 2025-06-10:
Also found the same issue, I submitted a fix

PR #361 (2025-06-11): Fix CameraInfo distortion coefficients and logger (backport #360)
The `camera_info_manager_py` packages uses the wrong distortion coefficients in the `CameraInfo` message. Additionally `rclpy.get_logger` should be `rclpy.logging.get_logger`.

Fix #354 <hr>This is an automatic backport of pull request #360 done by [Mergify](https://mergify.com).
PR #363 (2025-06-11): Fix CameraInfo distortion coefficients and logger (backport #360)
The `camera_info_manager_py` packages uses the wrong distortion coefficients in the `CameraInfo` message. Additionally `rclpy.get_logger` should be `rclpy.logging.get_logger`.

Fix #354 <hr>This is an automatic backport of pull request #360 done by [Mergify](https://mergify.com).
closed 2025-06-11 by ahcorde