ros-perception/image_common #355: republish 'node' functionality is broken in Jazzy

asimonov, 2025-04-29:
I summarised the problem with republish in Jazzy here: https://github.com/ros-perception/image_common/pull/275#issuecomment-2838833712

in that merge the 'node' was replaced by 'component', so any existing use of it as node now fails silently. not a great way to change API

mikeferguson, 2025-04-29:
I think you just need to change ```exec="republish"``` to ```exec="republish_node"``` in your launch file and your original will work.

There have been a number of updates in Jazzy. A fairly large, but not entirely comprehensive list is found in https://docs.ros.org/en/rolling/p/image_pipeline/changelog.html - it looks like this wasn't included, a PR to add that would certainly be merged.

peci1, 2026-01-16:
I think you were somehow confused. The `republish` node works exactly as it did before.

This is the list of installed files:

```
/opt/ros/jazzy/lib/image_transport
/opt/ros/jazzy/lib/image_transport/list_transports
/opt/ros/jazzy/lib/image_transport/republish
/opt/ros/jazzy/lib/libimage_transport.so
/opt/ros/jazzy/lib/libimage_transport_plugins.so
/opt/ros/jazzy/lib/librepublish_node.so
/opt/ros/jazzy/lib/x86_64-linux-gnu
/opt/ros/jazzy/lib/x86_64-linux-gnu/libimage_transport.so
```

You can clearly see the binary `/opt/ros/jazzy/lib/image_transport/republish` is there.

The binary is created from the component library using this utility macro:

```
rclcpp_components_register_node(republish_node
  PLUGIN "image_transport::Republisher"
  EXECUTABLE republish
)
```